arendst/Tasmota #1158: Bi directional REST API

smhgit, 2017-11-09:
ASFAIK you can send REST API to the sonoff but you cannot get information from sonoff via REST API.

Is there a plan to add REST API support from Sonoff device?

arendst, 2017-11-09:
?
Executing ``http://sonoff/cm?cmnd=status%202`` returns
``STATUS2 = {"StatusFWR":{"Version":"5.9.1b","BuildDateTime":"2017-11-08T16:18:36","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}``

Looks good to me.

smhgit, 2017-11-09:
@arendst yes, you get a response, but if you press a button on the sonoff, does it send the state vi REST API ?

davidelang, 2017-11-09:
so you are asking for the ability to send out messages via REST calls rather 
than just be able to retrieve status via REST calls.

I think that's going to get the same answer as the recent question about doing 
the same thing via HTTP calls, it would take too much code space to fit, so not 
an option.

you would need to fork the firmware and replace all MQTT functionality with your 
REST functionality. The refactoring would probably be a good thing, and that may 
result in something that can be at least a compile-time option.

But you will have to invent and define your REST protocol to replace the MQTT 
functionality.

David Lang

closed 2018-05-07 by stale[bot]