arendst/Tasmota #1159: No .bin file encryption

0n3-70uch, 2017-11-09:
Hello,

I saw that the .bin file isn't encrypted. It's easy to find the wifi and mqtt password in this file. A bit dangerous if the file is uploaded to a server (for upgrades).

Greetings
Fabi
[image]

arendst, 2017-11-09:
These parameters are stored in flash memory. People are overrating the possibility of editing the user_config.h file. No need to do this. 

Not editting the user_config.h file but using the appropriate commands the setting you see won't be the real values as long as you do not change the CFG_HOLDER define the flash settings are not overwritten.

0n3-70uch, 2017-11-09:
Ok so its better to upload a stock file and than change everything via mqtt or webinterface. I put some settings into the user_config_override.h because sometime I use the command "Reset 2". After a reset the sonoff need a working wifi configuration from the "user_config.h" file. Is there any other solution?

davidelang, 2017-11-09:
If you can compile your own .bin file and make it available on a webserver, you 
should also be able to add restrictions so that it can only be accessed by the 
things that you want to use it, and who need the passwords

0n3-70uch, 2017-11-10:
That's a good idea but therefore I need a password protection at the server side. How I can add the password string to the Sonoffs?

closed 2018-05-07 by stale[bot]