arendst/Tasmota #1160: Power Reading are sent over mqtt only every 5 minutes

SimoneBnc, 2017-11-09:
Hi, I there a way to send instant Power readings from Sonoff POW as soon as they are updated like on the web interface? I need to have it updates every few seconds and not 5 minutes. In the Domoticz section I can change the update time but it only changes how often the switch/relay status is updated. 
Thanks

arendst, 2017-11-09:
Change TelePeriod to as low as 10 seconds to receive more readings

SimoneBnc, 2017-11-09:
I did but the Domoticz publishing is still 5 minutes or am I doing it wrong?

edit21, 2017-11-10:
"TelePeriod 10" to set to ten seconds, check console for the messages being placed on to  mqtt - if they appear there every ten seconds or there abouts you have an issue on the Domoticz side - tail your logs.

SimoneBnc, 2017-11-11:
That did the trick. I modified "#define TELE_PERIOD 10" in sketch but it was still reporting every 5 minutes, but it was probably due to the fact that the first time I uploaded the sketch it was set to 300 and I guess there is this "hidden" setting that is saved in flash since now I still get reports every 10 seconds even after reboot. Thanks

closed 2017-11-11 by SimoneBnc