arendst/Tasmota #1161: Connect Sonoff RF Bridge Buttons to Domoticz

OliverHi, 2017-11-09:
I have recently flashed a Sonoff RF Bridge with tasmota. I have also already let it learn some button codes (I have a RF based switch I want to integrate in my home automation). These button presses show up on the MQTT broker.

Now my question is how do I get that connected to a button in domoticz (only incoming, I don't want to send anything)? The domoticz settings in my tasmota bridge only seem to show id(x) fields for potential connected sensors.

OliverHi, 2017-11-12:
PS Imho the 16 slots for learned RF codes could just be integrated like any other sonoff switch already can (creating virtual switches in domoticz setting the idx in tasmota).  A generic version is not really needed (at least for me).

tukestolo, 2018-01-05:
Friend could you figure it out ?*
I am try to connect rf bridge with Domoticz but I cannot do it :(

OliverHi, 2018-01-08:
No I did not. I am pretty sure this is not possible in Domoticz without Tasmota adding some code for it. For now I am using the RF Bridge with openhab which has a way more flexible MQTT binding.
I would still love support for this in Domoticz though as I am using this system in another install too.

tukestolo, 2018-01-08:
I found a way to use it to turn on and off switch’s.
In Domoticz you have the option to add a action on switch on and switch off. 

In that action you could put the url with the command to you rfbridge key that you what to send.
On wall switch you can défine on frequency to on and another to off.. I hope I could help you in something

Beliar-666, 2018-01-08:
Tukestolo, is it also possible to get the status back, if the channel is on or off? In Case i switch with the RF Remote?

tukestolo, 2018-01-08:
I don’t know... for now I believe that is not possible ... we need to explore to understand if when switch change state is possible to send something ...