arendst/Tasmota #1162: After Mqtt "cmnd" command missing data in "state" acknowledgement.

pcopa, 2017-11-09:
Has anyone observed this?

After OTA updating my modules to 5,9.0b I noticed that the module was behaving different. 
It seems that the prefix "oh/" that I added to the mqtt conf "oh/%prefix%/%topic%/" is not being transmitted in the "stat" acknowledgement anymore.
When pushing the button on the module it **is** transmitted!
See Below

[image]

[image]

ionciubotaru, 2017-11-10:
Set `FullTopic %prefix%/%topic%/`
If the problem still persist check the value of `prefix`

daballiemo, 2017-11-10:
same problem here:

user_config.h
`#define MQTT_FULLTOPIC         "home/esp8266/tasmota/%topic%/"`
`// %prefix% token options
#define SUB_PREFIX             "cmnd"            // [Prefix1] Sonoff devices subscribe to %prefix%/%topic% being SUB_PREFIX/MQTT_TOPIC and SUB_PREFIX/MQTT_GRPTOPIC
#define PUB_PREFIX             "stat"            // [Prefix2] Sonoff devices publish to %prefix%/%topic% being PUB_PREFIX/MQTT_TOPIC
#define PUB_PREFIX2            "tele"            // [Prefix3] Sonoff devices publish telemetry data to %prefix%/%topic% being PUB_PREFIX2/MQTT_TOPIC/UPTIME, POWER and TIME`

part of result:
`10:57:53 MQT: stat/sonoff_077697/STATUS = {"Status":{"Module":1, "FriendlyName":"sonoff_077697 Printer/Lader Stud", "Topic":"sonoff_077697", "ButtonTopic":"0", "Power":1, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}`

I would expect both SUB and PUB to follow MQTT_FULLTOPIC but it shows that SUB does follow MQTT_FULLTOPIC wher PUB only uses "PUB_PREFIX"

arendst, 2017-11-10:
@pcopa if you set the fallbacktopic (being mqttClient) equal to Mqtt topic you may expect this result as the latter two are provided using the fallbacktopic and is as designed. Fallback topic is just that: a fallback in case you forget the Mqtt topic and the result will always NOT use the full topic.

@daballiemo looks the same situation.

Bottomline: Use different names for MQTTClient and MQTT topic.

daballiemo, 2017-11-10:
@arendst,

thx for your quick respons. I have changed my setup to what I think I understand in your response. I however still have one issue, when firing a statusupdate for among others the firmware version;
using the same build (5.9.1.a)

Sonoff Basic:
`16:51:28 RSL: Received Topic home/esp8266/tasmota/sonoff_2A16BC/cmnd/status, Data Size 1, Data 2
16:51:28 RSL: Group 0, Index 1, Command STATUS, Data 2
16:51:29 MQT: home/esp8266/tasmota/sonoff_2A16BC/STATUS2 = {"StatusFWR":{"Version":"5.9.1a", "BuildDateTime":"2017-11-10T16:41:02", "Boot":5, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}`

H801:
`16:51:29 RSL: Received Topic home/esp8266/tasmota/sonoff_D65C77/cmnd/status, Data Size 1, Data 2
16:51:29 RSL: Group 0, Index 1, Command STATUS, Data 2
16:51:29 MQT: stat/sonoff_D65C77/STATUS2 = {"StatusFWR":{"Version":"5.9.1a", "BuildDateTime":"2017-11-10T16:41:02", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}`

what can be seen is that it looks like the MQTT respons of the H801 does not respect the settings in user_config.h even not after reset:
`17:01:55 RSL: Received Topic /Fulltopic, Data Size 1, Data 1
17:01:55 RSL: Group 0, Index 1, Command FULLTOPIC, Data 1
17:01:55 MQT: home/esp8266/tasmota/sonoff_D65C77/stat/RESULT = {"FullTopic":"home/esp8266/tasmota/%topic%/%prefix%/"}
17:02:00 MQT: home/esp8266/tasmota/sonoff_D65C77/tele/UPTIME = {"Time":"2017-11-10T17:02:00", "Uptime":1}
17:02:13 WIF: Checking connection...
17:02:13 WIF: Connected
17:02:33 WIF: Checking connection...
17:02:33 WIF: Connected
17:02:53 WIF: Checking connection...
17:02:53 WIF: Connected
17:03:13 WIF: Checking connection...
17:03:13 WIF: Connected
17:03:33 WIF: Checking connection...
17:03:33 WIF: Connected
17:03:36 RSL: Received Topic home/esp8266/tasmota/sonoff_D65C77/cmnd/status, Data Size 1, Data 2
17:03:36 RSL: Group 0, Index 1, Command STATUS, Data 2
17:03:36 MQT: stat/sonoff_D65C77/STATUS2 = {"StatusFWR":{"Version":"5.9.1a", "BuildDateTime":"2017-11-10T16:41:02", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}`

arendst, 2017-11-10:
Check your MQTTClient and Topic for the H801. Apparently they are still the same.

Use commands ``mqttclient`` and ``topic`` to check and/or update.

pcopa, 2017-11-10:
@ Thanks for the prompt response. I'm new to MQTT and will need some more time  to dig into the fallback topic before I get back to you.
I'm using HomeAssistant and was confused because I've been doing OTA updates for 2 months with the same configuration (full topic) without any issues and suddenly the behavior changed.

daballiemo, 2017-11-14:
I have changed my setup as instructed and everything is now working as it should.

thx

pcopa, 2017-11-15:
I've also changed my configuration for all my modules and now everything is working fine.
What I have learned, (with the help of @arendst), is that the "Configuration->Configure Other->Friendly Name 1 = "xyz" name and the name in the topic .../.../xyz/... can **NOT** be the same. 
In my case I have added the prefix "m" (for module) to all my names (**m**xyz) in all the mqtt topics.

closed 2018-05-07 by stale[bot]