arendst/Tasmota #1164: Add sonoff to HomeAssistant project.

shayshtern, 2017-11-10:
Can someone explain me exactly how to add sonoff that formated with Tasmota project and have web access to HomeAssistant project ?
( http://hassio.local:8123 ). 
I really can't get it. Thank you! 
Thanks !

ionciubotaru, 2017-11-10:
Read https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

shayshtern, 2017-11-10:
Yea I read it. But can't really understant the code. 
Any other maybe more explaining tutorial ?

gajotnt, 2017-11-10:
Use domoticz :P

shayshtern, 2017-11-10:
What do you mean?

gajotnt, 2017-11-10:
Use domoticz instead of Home Assistant, way simpler to configure and use.

shayshtern, 2017-11-10:
Supported homekit and Google home?!

gajotnt, 2017-11-10:
HomeKit yes
https://www.domoticz.com/wiki/Homekit_Siri

Google Home I dont think its quite implemented yet, and i find that wierd

shayshtern, 2017-11-10:
Homekit for iPhone and some : hi Google turn on lights" on Android. How do I do that? Is there "Home" app for Android?

closed 2018-05-07 by stale[bot]