arendst/Tasmota #1165: Supporting Instructables

MikePRoberts, 2017-11-10:
I have published two instructables that I hope help and support what you are doing.  If you feel I have missed anything or should make any change to these please let me know.  The first details setting up Node-Red and mosquitto on a Raspberry PI and initial control of a Sonoff Switch:  
https://www.instructables.com/id/Powerful-Standalone-Home-Automation-System-Pi-Sono/
The second shows examples of sensors and LEDs using Sonoff-Tasmota:
https://www.instructables.com/id/Home-Automation-Sonoff-Tasmota-Sensors-LEDs-Develo/
Thank you for your great work here.
Mike

arendst, 2017-11-10:
@MikePRoberts thnx for the instructables. Always nice to see how people expand on my groundwork.

ascillato, 2018-04-23:
Added to the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/NodeRed)

Can be closed now

closed 2018-06-02 by ascillato