arendst/Tasmota #1168: Able to broadcast state of relay(s)?

TommySharpNZ, 2017-11-11:
I have a few sonoff basics and 4ch devices.

I have various ways of sending MQTT commands to turn them on and off. 
An issue is where my main home automation does not see the MQTT message and then does not reflect the current state of the POW.

Most of the time this is not an issue but is there any way to broadcast the state of the relay say once per hour so all other systems are fully aware of the state?

arendst, 2017-11-11:
Currently only possible by pulling it in using command ``status 11``

davidelang, 2017-11-12:
don't relay states also get sent as part of the tele/ messages?
k

arendst, 2017-11-12:
They are sent indeed without every tele message.

TommySharpNZ, 2017-11-12:
Thanks guys, how often would a Basic or 4CH sonoff send out a tele message?

davidelang, 2017-11-13:
On Sun, 12 Nov 2017, TommySharpNZ wrote:

> Thanks guys, how often would a Basic or 4CH sonoff send out a tele message?

the default is every 5 min, you can set it as short as every 10 seconds

stefanbode, 2017-11-16:
`"teleperiod 10"` 
is the magic to get it continuously. I also use this trick to ensure to get the right state. This is quite convenient to ensure the matching status is reported. I also use the STATUS to update the LAST connect string. Therefore not only the LWT does show if the device is connected. I also can see, when there was an update send.

closed 2018-05-07 by stale[bot]