arendst/Tasmota #1169: not working when network down

kezzab25, 2017-11-11:
I am not sure if my config is wrong but how do I get a sonoff basic to work when the network goes down? Unfortunately that happens a lot here as our router is in the main house and the owner likes to disconnect me from the network often.

freaktech, 2017-11-11:
hi

i have the TOUCH and tested NOW

it is working the most times but def. sometimes not

sometimes the LED for WIFI starts flashing, then it will be ON   (also when there is no WIFI)

so its not working 100% when WIFI is off   in 5,9

tom

davidelang, 2017-11-12:
When the network is down, the device will attempt to reconnect to the network 
and to the mqtt server. While it is attempting to connect to the mqtt server, 
the library prevents anything else from happening (including responding to local 
buttons), so you will see 3-5 second delays on a regular basis.

run your own router and Pi with your mqtt server and home automation server. 
Then have your router connect to the main house's router for connections out. 
When the Internet connection goes down, all your local stuff will keep working.
k

freaktech, 2017-11-12:
i don't use MQTT so its only a problem if the router would be down

davidelang, 2017-11-13:
disable mqtt if you aren't using it and you will not run into the delays of the 
device attempting to connect to the mqtt server.

David Lang

stefanbode, 2017-11-16:
Is there a config option or compiler option to influence the timeout for the MQTT server connection. Normally the server answers quite fast. Therefore a hardcoded change of the timeout can be something we can take into consideration.

davidelang, 2017-11-17:
This is a bug somewhere in the arduino network library, we havent' gone digging 
to find out exactly where and what it would take to override it

> Is there a config option or compiler option to influence the timeout for the MQTT server connection. Normally the server answers quite fast. Therefore a hardcoded change of the timeout can be something we can take into consideration.
>
>

closed 2018-05-07 by stale[bot]