arendst/Tasmota #1170: Sonoff T1 with new 5.9.1c - not booting

Wieg, 2017-11-11:
Hi Guys,

Just flashed a T1 UK and the flash went ok but the device doesn't startup.

It is an esp8285 i only changed the default wifi SSID and password and flashed it with DOUT.
So i am a bit stuck, what to do now ? any thoughts are very appreciated.

jeylites, 2017-11-11:
Have you check the flash size? Did you flag using SonOTA or manually with GPIO 0 to Ground?

Wieg, 2017-11-11:
flash size was OK flash was SUCCES i did it manually GPIO 0 to ground. 
I don't get .....

jeylites, 2017-11-11:
Don't know if would it make any difference, have you tried V.5.8 also what are you getting on serial terminal, any messages?

Wieg, 2017-11-11:
got one working with 5.8p, the first one i couldn't even get in flashmode anymore.
Hope i didn't brick it ........ :(

jeylites, 2017-11-11:
The one you can't get into flash mode, was it flashed with 5.9 before?

*Jey Kasi*

Wieg, 2017-11-11:
no fresh out the box ........ i tried it for 30 minutes now ..... can't get it to go in flash anymore .... but it seems not to be a brick i think / hope

jeylites, 2017-11-11:
Don't know if it's a related problem but I will share my past experinece. My very first attempt to flash the T1 was through SonOTA and it failed so manual upload was next. Even then doing it manually, I couldn't get it into flash mode.  By accident, I discovered the Atmel (Marked in Yellow) somehow was interfering with the ESP so I removed it and was able to flash the ESP without any problem.

[image]

Wieg, 2017-11-11:
the second it did out off the box with the 5.8p not the 5.9.1c (thats all i can think off for now) .... but .... my package arrived today .... just did 2 sonoff pow's out off the box with the 5.9.1c and they work like a charm ........

Wieg, 2017-11-12:
Just could not leave it alone .... i just flashed my "bricked" T1 UK .... and bam in one time it worked .... just call me a noob .... did the same as always ... now it works.

Now the lights and devices in my house are allmost all sonoff based :) 29 devices and counting :)
big thanks to @arendst :)

closed 2018-05-07 by stale[bot]