arendst/Tasmota #1171: Tasmota-MQTT-Domoticz, multiple questions

crispy78, 2017-11-11:
First of all I would like to thank the Tasmota community for making such a wonderful product.
I've got serveral Sonoff switches installed in my home (Sonoff S20, Sonoff Basic, Sonoff POW) and with Tasmota and Domoticz I'm guite happy with my Domotica-setup.

I've seen the "Update Timer (0)" in Configuration -> Configure Domoticz, but I can't find out what it does or what it's purpose is.

I've read that @smadds publishes the version-number of Tasmota every 10 minutes on sonoff.maddox.co.uk:1883 with Topic sonoff-version. Can anyone tell me how to implement it on my Domoticz installation? I would like Domoticz to send an update command to all my devices when the version-number changes.

arendst, 2017-11-12:
Domoticz update timer just sends the power status only at that interval to Domoticz. I do not use it.

smadds, 2017-11-12:
Sorry, I don't use Domoticz. Some node red examples below if they help.

To clarify, the MQTT version is not updated every 10mins, but only when the release version changes. I look for a new version on Github every 10 minutes, make local copies available via http then publish the new version to the MQTT broker.

**Example Node-Red flows for Tasmota upgrades**

>` [{"id":"72c9d1c2.d026d","type":"mqtt in","z":"e41cf9a3.0f0538","name":"","topic":"sonoff-version","qos":"2","broker":"d81484a8.438f38","x":190,"y":1420,"wires":[["4ff9a647.03c1a8","10f7372b.850609"]]},{"id":"4ff9a647.03c1a8","type":"ui_text","z":"e41cf9a3.0f0538","group":"b66688fa.da3ab8","order":1,"width":0,"height":0,"name":"","label":"Available Version","format":"{{msg.payload}}","layout":"row-spread","x":410,"y":1400,"wires":[]},{"id":"9246f1c3.f4a25","type":"ui_button","z":"e41cf9a3.0f0538","name":"Upgrade Devices","group":"","order":4,"width":"0","height":"0","passthru":false,"label":"Upgrade Devices","color":"","bgcolor":"","icon":"","payload":"sonoff-version","payloadType":"global","topic":"cmnd/sonoffs/upgrade","x":210,"y":1540,"wires":[["a28deb3d.66aeb8"]]},{"id":"a28deb3d.66aeb8","type":"mqtt out","z":"e41cf9a3.0f0538","name":"","topic":"","qos":"","retain":"","broker":"","x":430,"y":1540,"wires":[]},{"id":"10f7372b.850609","type":"function","z":"e41cf9a3.0f0538","name":"Set global","func":"global.set(\"sonoff-version\",msg.payload.substr(1))\nreturn msg;","outputs":1,"noerr":0,"x":380,"y":1440,"wires":[[]]},{"id":"57232b66.786a64","type":"comment","z":"e41cf9a3.0f0538","name":"Scan Devices","info":"","x":170,"y":1600,"wires":[]},{"id":"b1a6d9f6.c905d8","type":"ui_button","z":"e41cf9a3.0f0538","name":"Scan Devices","group":"","order":2,"width":"0","height":"0","passthru":false,"label":"Scan Devices","color":"","bgcolor":"","icon":"","payload":"0","payloadType":"str","topic":"cmnd/sonoffs/status","x":200,"y":1700,"wires":[["486fcd33.554e74","753adfb8.4ed0a"]]},{"id":"753adfb8.4ed0a","type":"mqtt out","z":"e41cf9a3.0f0538","name":"","topic":"","qos":"","retain":"","broker":"","x":390,"y":1640,"wires":[]},{"id":"c720b5ff.0b1c88","type":"inject","z":"e41cf9a3.0f0538","name":"Hourly Scan","topic":"cmnd/sonoffs/status","payload":"2","payloadType":"num","repeat":"3600","crontab":"","once":true,"x":200,"y":1640,"wires":[["486fcd33.554e74","753adfb8.4ed0a"]]},{"id":"6d6038c5.d500a8","type":"ui_text","z":"e41cf9a3.0f0538","group":"","order":3,"width":"6","height":"1","name":"","label":"Last Scanned","format":"{{msg.timestamp | date:\"EEE d MMM, HH:mm\"}}","layout":"row-spread","x":620,"y":1700,"wires":[]},{"id":"486fcd33.554e74","type":"function","z":"e41cf9a3.0f0538","name":"Add timestamp","func":"if ( !msg.timestamp ) msg.timestamp = +new Date();\nreturn msg;","outputs":1,"noerr":0,"x":420,"y":1700,"wires":[["6d6038c5.d500a8"]]},{"id":"42a9619d.e2b95","type":"comment","z":"e41cf9a3.0f0538","name":"Get Firmware Version","info":"","x":200,"y":1380,"wires":[]},{"id":"a0914f04.b3f99","type":"comment","z":"e41cf9a3.0f0538","name":"Manual Upgrade","info":"","x":180,"y":1500,"wires":[]},{"id":"8a1c3db5.bb023","type":"mqtt in","z":"e41cf9a3.0f0538","name":"","topic":"stat/**device-name**/STATUS2","qos":"2","broker":"","x":250,"y":1800,"wires":[["ac3c0f3a.84247"]]},{"id":"5076a666.0b6958","type":"ui_text","z":"e41cf9a3.0f0538","group":"","order":8,"width":0,"height":0,"name":"**Device Name** version","label":"","format":"v{{msg.payload.StatusFWR.Version}}","layout":"row-spread","x":670,"y":1800,"wires":[]},{"id":"ac3c0f3a.84247","type":"json","z":"e41cf9a3.0f0538","name":"","pretty":false,"x":470,"y":1800,"wires":[["5076a666.0b6958"]]},{"id":"26412ee9.b60d72","type":"comment","z":"e41cf9a3.0f0538","name":"Repeat below for each device","info":"","x":220,"y":1760,"wires":[]},{"id":"d81484a8.438f38","type":"mqtt-broker","z":"","broker":"sonoff.maddox.co.uk","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""},{"id":"b66688fa.da3ab8","type":"ui_group","z":"","name":"Upgrade","tab":"a73bba8f.a27df8","order":4,"disp":true,"width":"6"},{"id":"a73bba8f.a27df8","type":"ui_tab","z":"","name":"Settings","icon":"dashboard","order":2}]`

closed 2018-05-07 by stale[bot]