arendst/Tasmota #1172: Physical push button with existing DHT22 sensor?

C-Duv, 2017-11-12:
Hello,
I have an already working setup of a Sonoff Basic running Tasmota (v5.2.4) and reading temperature and humidity on a DHT22 connected on GPIO14 (in *Module* settings page, *GPI014 Sensor* is set to *03 DHT22*). Both relay state and DHT22 values are retrieved on home automation software via MQTT).

I would like to add a push (momentary) button to this setup so that pushing the button toggles the relay (directly but also sends the new state MQTT message).
I know I'll have to configure the [SwitchMode and/or SwitchTopic](https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic) but I am more concerned about how and where connect the push button...

I have found issue #508 (*Toggle Switch on GPIO 1(RX) or 3(TX)*) which suggest to use *GPIO03* (*RX*) but it uses a **toggle**/**maintained** switch not a **push**/**momentary** like I want, so I am not sure the following schema works:
[image]

*(Ignore the gray "F" zones: they symbolizes female pins on a male header I soldered on the Sonoff board)*

ionciubotaru, 2017-11-12:
Yes, your schema is OK and must work fine
If you receive false triggers from push button add a small cap between gpio3 and gnd

smadds, 2017-11-12:
100nF across the switch will help debounce the switch

Rene-Auberger, 2017-11-14:
@smadds do you mean the capacitor will be connected to GND and GPIO3/VCC in parallel to the switch? Or where is it located exactly (sorry, I'm a software guy ;-) and currently refreshing my electronics skills).

smadds, 2017-11-14:
It can be either end of the wire - it's just easier to mount it on the back of the switch. Effectively it's between the I/O pin and gnd

C-Duv, 2017-11-15:
@ionciubotaru @smadds If I get this correct: I should first try with the 4.7k resistor only.
Then, if I get false triggers, I should add the 100nF capacitor (**keeping the resistor**).

2e0vet, 2017-11-19:
I dont suppose you have documented how you get the temp readings, I have been trying for ages to get this working without success

C-Duv, 2017-11-26:
@2e0vet I used a AM2302 sensor connected as follows on the Sonoff:

**AM2302 pins:**

| Pin number | Usage        |
|------------|--------------|
| 1          | VCC (Power)  |
| 2          | SDA (Data)   |
| 3          | *Nothing*    |
| 4          | GND (Ground) |

(Pins are numbered from left to right: see [this specification document](http://akizukidenshi.com/download/ds/aosong/AM2302.pdf)).

**Connection to the Sonoff:**

| DHT22/AM2302 | Sonoff |
|--------------|--------|
| VCC (pin 1)  | 3V3    |
| SDA (pin 2)  | GPIO14 |
| GND (pin 4)  | GND    |

(See [GPIO Locations](https://github.com/arendst/Sonoff-Tasmota/wiki/GPIO-Locations) for Sonoff pins locations)

Then of the Tasmota web interface, go to *Configuration* > *Configure Module* and set the *GPI014 Sensor* field to *03 DHT22*. Click the *Save* button and the Sonoff should reboot. After a few seconds (~30s) temperatures and readings should be displayed on the web interface main page.

C-Duv, 2017-11-29:
My setup (only using a 4.7kΩ resistor) works fine.

In *Configuration* > *Configure Module* page, I set *GPIO3 Serial In* to the value *09 Switch1*.
And changed the *SwitchMode* to 3 by accessing the following URL:
`curl http://THE.SONOFF.IP.ADDRESS/cm?cmnd=SwitchMode1%203`
*(`SwitchMode1%203` = `SwitchMode1 3` = Set SwitchMode for relay #1 to mode 3)*

closed 2017-11-29 by C-Duv