arendst/Tasmota #1174: Failover based on mqtt heartbeat timeout - mission critical control?

RobertSzabo, 2017-11-12:
Hi,

First of all, great firmware!!! Thanks!
I wonder, if there is anything close to provide failover (e.g., device restart) when publication to an mqtt topic is missing for a given amount of time. 

Use-case:
Intelligent thermostat: Call For Heat is controlled via a switch; assume it is ON but communication with the mqtt server or the thermostat controller is lost. After a given timeout, the switch (device) should reboot. Note: config is set to default switch to OFF on reboot.  

What do you think?
Br,
/r

davidelang, 2017-11-13:
On Sun, 12 Nov 2017, Robert wrote:

> First of all, great firmware!!! Thanks!
> I wonder, if there is anything close to provide failover (e.g., device restart) when publication to an mqtt topic is missing for a given amount of time.

not device restart, but reconnect attempts or connecting to a different system.
kk

RobertSzabo, 2017-11-13:
> not device restart, but reconnect attempts or connecting to a different system.
different use-case, which is already supported. 
Device A should do a failover if Device B heartbeats are missing for a given amount of time. 
I hope it clarifies.

closed 2018-05-07 by stale[bot]