arendst/Tasmota #1175: Sonoff SC not reporting sensor data

EmmanuelLM, 2017-11-12:
Hi, just flashed 5.9.1 on my sonoff sc.
The device connects to the wifi and mqtt broker with no issues but it never reports any sensor state...
the console is as follows:

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_F8B16C, GroupTopic sonoffs) Version 5.9.1
00:00:00 WIF: Connecting to AP1 beancave in mode 11N as sonoff-4460...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on sonoff-4460.local with IP address 192.168.1.121
20:28:09 MQT: Attempting connection...
20:28:09 MQT: Connect failed to 192.168.1.139:1883, rc -2. Retry in 10 sec
20:28:20 MQT: Attempting connection...
20:28:29 MQT: Connected
20:28:30 MQT: tele/sonoff/LWT = Online (retained)
20:28:30 MQT: cmnd/sonoff/POWER = 
20:28:30 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff SC", "Version":"5.9.1", "FallbackTopic":"DVES_F8B16C", "GroupTopic":"sonoffs"}
20:28:31 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin", "Hostname":"sonoff-4460", "IPAddress":"192.168.1.121"}
20:28:31 MQT: tele/sonoff/INFO3 = {"RestartReason":"Power on"}
20:28:39 MQT: tele/sonoff/STATE = {"Time":"2017-11-12T20:28:39", "Uptime":0, "Vcc":3.192, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
20:29:40 MQT: tele/sonoff/STATE = {"Time":"2017-11-12T20:29:39", "Uptime":0, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
20:30:43 MQT: tele/sonoff/STATE = {"Time":"2017-11-12T20:30:43", "Uptime":0, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
20:31:51 MQT: tele/sonoff/STATE = {"Time":"2017-11-12T20:31:51", "Uptime":0, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
20:32:58 MQT: tele/sonoff/STATE = {"Time":"2017-11-12T20:32:58", "Uptime":0, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
20:34:03 MQT: tele/sonoff/STATE = {"Time":"2017-11-12T20:34:03", "Uptime":0, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}

Any idea what I'm missing??

arendst, 2017-11-12:
Try a powercycle to get the atmel processor started.

EmmanuelLM, 2017-11-12:
I tried that, unplugged it and plugged it back in

arendst, 2017-11-12:
If you uploaded the firmware via serial you did not forget to reinsert the serial jumpers?

Enable more logging to see atmel communication (weblog 3). Do not use seriallog as it uses serial to communicate with atmel.

EmmanuelLM, 2017-11-12:
Did that, so far I cannot see any Comms... I will open it up tomorrow to check the tx jumper but 99% certain I put it back!

22:24:49 LOG: SerialLog 0, WebLog 3, SysLog 1, LogHost domus1, LogPort 514, TelePeriod 60
22:24:49 HTP: Configuration
22:24:50 CFG: Saved to flash at F4, Count 32, Bytes 1648
22:25:04 HTP: Main Menu
22:25:09 HTP: Console
22:25:37 MQT: tele/sonoff/STATE = {"Time":"2017-11-12T22:25:37", "Uptime":2, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
22:26:44 MQT: tele/sonoff/STATE = {"Time":"2017-11-12T22:26:44", "Uptime":2, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
22:27:49 MQT: tele/sonoff/STATE = {"Time":"2017-11-12T22:27:49", "Uptime":2, "Vcc":3.191, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
22:28:59 MQT: tele/sonoff/STATE = {"Time":"2017-11-12T22:28:59", "Uptime":2, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}

EmmanuelLM, 2017-11-14:
I checked this morning and the jumper is in. I have also changed the power supply and the usb cable (just in case) but no luck :-( any help appreciated!

09:05:35 MQT: Connected
09:05:35 MQT: tele/sonoff/LWT = Online (retained)
09:05:35 MQT: cmnd/sonoff/POWER = 
09:05:35 MQT: Subscribe to cmnd/sonoff/#
09:05:35 MQT: Subscribe to cmnd/sonoffs/#
09:05:35 MQT: Subscribe to cmnd/DVES_F8B16C/#
09:05:36 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff SC", "Version":"5.9.1", "FallbackTopic":"DVES_F8B16C", "GroupTopic":"sonoffs"}
09:05:36 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin", "Hostname":"sonoff-4460", "IPAddress":"192.168.1.121"}
09:05:36 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
09:05:44 MQT: tele/sonoff/STATE = {"Time":"2017-11-14T09:05:44", "Uptime":0, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
09:05:47 HTP: Main Menu
09:06:38 HTP: Console
09:06:50 MQT: tele/sonoff/STATE = {"Time":"2017-11-14T09:06:50", "Uptime":0, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
09:07:58 MQT: tele/sonoff/STATE = {"Time":"2017-11-14T09:07:58", "Uptime":0, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
09:08:59 MQT: tele/sonoff/STATE = {"Time":"2017-11-14T09:08:59", "Uptime":0, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
09:09:59 MQT: tele/sonoff/STATE = {"Time":"2017-11-14T09:09:59", "Uptime":0, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
09:11:00 MQT: tele/sonoff/STATE = {"Time":"2017-11-14T09:11:00", "Uptime":0, "Vcc":3.190, "Wifi":{"AP":1, "SSId":"beancave", "RSSI":100, "APMac":"9C:3D:CF:06:48:D9"}}
09:11:18 HTP: Main Menu
09:11:20 HTP: Console

EmmanuelLM, 2017-11-14:
I don't know if it's relevant but my PCB number is v2.0...
I will try and flash the Atmel chip but I suspect I should see traffic from the ESP to the Atmel at minima, even if the Atmel does not respond, no?

arendst, 2017-11-14:
With weblog 3 you should see traffic from both Atmel and esp8266 like:
```
14:01:18 SER: Transmit AT+SEND=ok
14:01:25 SER: Received AT+UPDATE="humidity":15,"temperature":24,"light":10,"noise":3,"dusty":1
14:01:25 SER: Transmit AT+SEND=ok
14:01:30 SER: Received AT+UPDATE="humidity":15,"temperature":24,"light":10,"noise":2,"dusty":1
14:01:30 SER: Transmit AT+SEND=ok
14:01:33 SER: Received AT+UPDATE="humidity":15,"temperature":24,"light":10,"noise":3,"dusty":1
14:01:33 SER: Transmit AT+SEND=ok
14:01:43 SER: Received AT+STATUS?
14:01:43 SER: Transmit AT+STATUS=4
14:01:50 SER: Received AT+UPDATE="humidity":15,"temperature":25,"light":10,"noise":2,"dusty":1
14:01:50 SER: Transmit AT+SEND=ok
14:01:54 SER: Received AT+UPDATE="humidity":15,"temperature":24,"light":10,"noise":2,"dusty":1
14:01:54 SER: Transmit AT+SEND=ok
```

EmmanuelLM, 2017-11-14:
yeah, I'm not seeing any of this... which is odd because I would expect to at least see the "Transmit" lines even if the Atmel doesn't answer....

Will try reflashing everything

zibous, 2017-11-23:
Hi Theo,

Same problem here, Sonoff SC do not show any data and do not report data
as mqtt topic.
Only these messages i can see:

```
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on IPS-SC101.local with IP address 10.1.1.101
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/THC101/LWT = Online (retained)
00:00:06 MQT: cmnd/THC101/POWER = 
00:00:06 MQT: Subscribe to cmnd/THC101/#
00:00:06 MQT: Subscribe to cmnd/sonoffs/#
00:00:06 MQT: Subscribe to cmnd/SC-THC101/#
00:00:06 MQT: tele/THC101/INFO1 = {"Module":"Sonoff SC","Version":"5.9.1h","FallbackTopic":"SC-THC101","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/THC101/INFO2 = {"WebServerMode":"Admin","Hostname":"IPS-SC101","IPAddress":"10.1.1.101"}
00:00:06 MQT: tele/THC101/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 HTP: Main Menu
00:00:06 APP: (UTC) Thu Nov 23 14:19:16 2017
00:00:06 APP: (DST) Sun Mar 26 02:00:00 2017
00:00:06 APP: (STD) Sun Oct 29 03:00:00 2017
16:19:18 HTP: Console
16:19:21 SER: Received AT+STATUS?
16:19:21 SER: Transmit AT+STATUS=4
16:19:24 MQT: tele/THC101/STATE = {"Time":"2017-11-23T16:19:24","Uptime":0,"Vcc":3.164,"Wifi":{"AP":1,"SSId":"McWille1","RSSI":100,"APMac":"10:9A:DD:88:34:65"}}
```
[image]

Program Version | 5.9.1h
-- | --

arendst, 2017-11-23:
Pls provide output of command status 0.

zibous, 2017-11-24:
Works now, sorry and thanks for your great job.

```
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on IPS-SC101.local with IP address 10.1.1.101
00:00:05 SER: Received AT+UPDATE="humidity":14,"temperature":20,"light":8,"noise":1,"dusty":1
00:00:05 SER: Transmit AT+SEND=ok
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/THC101/LWT = Online (retained)
00:00:06 MQT: cmnd/THC101/POWER = 
00:00:06 MQT: Subscribe to cmnd/THC101/#
00:00:06 MQT: Subscribe to cmnd/sonoffs/#
00:00:06 MQT: Subscribe to cmnd/SC-THC101/#
00:00:06 MQT: tele/THC101/INFO1 = {"Module":"Sonoff SC","Version":"5.9.1h","FallbackTopic":"SC-THC101","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/THC101/INFO2 = {"WebServerMode":"Admin","Hostname":"IPS-SC101","IPAddress":"10.1.1.101"}
00:00:06 MQT: tele/THC101/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 APP: (UTC) Fri Nov 24 08:12:50 2017
00:00:06 APP: (DST) Sun Mar 26 02:00:00 2017
00:00:06 APP: (STD) Sun Oct 29 03:00:00 2017
10:12:54 HTP: Main Menu
10:12:58 HTP: Console
10:12:58 MQT: tele/THC101/STATE = {"Time":"2017-11-24T10:12:58","Uptime":0,"Vcc":3.169,"Wifi":{"AP":1,"SSId":"McWille1","RSSI":86,"APMac":"10:9A:DD:88:34:65"}}
10:12:58 MQT: tele/THC101/SENSOR = {"Time":"2017-11-24T10:12:58","Temperature":20.00,"Humidity":14.00,"Light":30,"Noise":10,"AirQuality":100,"TempUnit":"C"}
10:12:59 SER: Received AT+STATUS?
10:12:59 SER: Transmit AT+STATUS=4
10:13:07 WIF: Checking connection...
10:13:07 WIF: Connected
10:13:09 SER: Received AT+STATUS?
10:13:09 SER: Transmit AT+STATUS=4
10:13:19 SER: Received AT+STATUS?
10:13:19 SER: Transmit AT+STATUS=4
10:13:27 WIF: Checking connection...
10:13:27 WIF: Connected
10:15:20 MQT: stat/THC101/STATUS = {"Status":{"Module":21,"FriendlyName":"IPS-SC001","Topic":"THC101","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:15:28 WIF: Checking connection...
10:15:28 WIF: Connected
10:15:29 SER: Received AT+STATUS?
10:15:29 SER: Transmit AT+STATUS=4
....

krasatos, 2017-12-23:
Didn't understand what fixed it.
Still having trouble getting dht22 data on sonoff basic.
here is my status 0

```
14:05:05 MQT: stat/sonoff2/STATUS = {"Status":{"Module":1, "FriendlyName":"test sonoff", "Topic":"sonoff2", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":0, "ButtonRetain":0, "PowerRetain":0}}
14:05:05 MQT: stat/sonoff2/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":1, "Sleep":100, "BootCount":64, "SaveCount":173, "SaveAddress":"F7000"}}
14:05:05 MQT: stat/sonoff2/STATUS2 = {"StatusFWR":{"Version":"5.9.1", "BuildDateTime":"2017-11-13T21:43:41", "Boot":6, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
14:05:05 MQT: stat/sonoff2/STATUS3 = {"StatusLOG":{"SerialLog":2, "WebLog":2, "SysLog":0, "LogHost":"domus1", "LogPort":514, "SSId1":"Paleo", "SSId2":"****", "TelePeriod":300, "SetOption":"55800008"}}
14:05:05 MQT: stat/sonoff2/STATUS4 = {"StatusMEM":{"ProgramSize":471, "Free":532, "Heap":25, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":3}}
14:05:05 MQT: stat/sonoff2/STATUS5 = {"StatusNET":{"Hostname":"sonoff2-8022", "IPAddress":"****", "Gateway":"****", "Subnetmask":"255.255.255.0", "DNSServer":"208.67.222.222", "Mac":"2C:3A:E8:0F:3F:56", "Webserver":2, "WifiConfig":2}}
14:05:05 MQT: stat/sonoff2/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.200", "MqttPort":1883, "MqttClientMask":"sonofflivingroom_%06X", "MqttClient":"sonofflivingroom_0F3F56", "MqttUser":"***", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
14:05:05 MQT: stat/sonoff2/STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 23 12:05:05 2017", "Local":"Sat Dec 23 14:05:05 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":2}}
14:05:05 MQT: stat/sonoff2/STATUS10 = {"StatusSNS":{"Time":"2017-12-23T14:05:05"}}
14:05:05 MQT: stat/sonoff2/STATUS11 = {"StatusSTS":{"Time":"2017-12-23T14:05:05", "Uptime":1, "Vcc":3.205, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"****", "RSSI":84, "APMac":"60:E3:27:35:8D:BE"}}}
```
(masked off some stuff with *** , hope im not disclosing something sensitive)

[image]

tlc76, 2019-07-11:
Hi, today I upgraded my Sonoff SC remotely via Firmware Upgrade / Upgrade by file upload, from Tasmota 6.1.1 to 6.5.0 (two step upgrade: 1. sonoff-minimal 2. sonoff / sonoff-sensors).
Finally I have configured the module properly (by selecting Sonoff-SC from the dropdown list).
In both cases, the SC is not reporting any sensor data.
The console says:

> 23:33:06 MQT: tasmota/sonoff-sc/stat/RESULT = {"WebLog":3}
23:33:07 CFG: Saved to flash at F9, Count 14, Bytes 3584
23:33:12 SER: Received AT+UPDATE="humidity":,"temperature":,"light":,"noise":,"dusty":
23:33:12 SER: Transmit AT+SEND=fail
23:33:22 SER: Received AT+STATUS?
23:33:22 SER: Transmit AT+STATUS=4
23:33:29 SER: Received AT+UPDATE="humidity":,"temperature":,"light":,"noise":,"dusty":
23:33:29 SER: Transmit AT+SEND=fail
23:33:39 SER: Received AT+STATUS?
23:33:39 SER: Transmit AT+STATUS=4
23:33:49 SER: Received AT+STATUS?
23:33:49 SER: Transmit AT+STATUS=4
23:34:01 SER: Received AT+UPDATE="humidity":,"temperature":,"light":,"noise":,"dusty":
23:34:01 SER: Transmit AT+SEND=fail
23:34:11 SER: Received AT+STATUS?
23:34:11 SER: Transmit AT+STATUS=4
23:34:15 SER: Received AT+UPDATE="humidity":,"temperature":,"light":,"noise":,"dusty":
23:34:15 SER: Transmit AT+SEND=fail
23:34:21 SER: Received AT+UPDATE="humidity":,"temperature":,"light":,"noise":,"dusty":
23:34:21 SER: Transmit AT+SEND=fail
23:34:28 MQT: tasmota/sonoff-sc/tele/STATE = {"Time":"2019-07-11T23:34:28","Uptime":"0T00:05:32","SleepMode":"Dynamic","Sleep":50,"LoadAvg":29,"Wifi":{"AP":1,"SSId":"tlc3","BSSId":"78:11:DC:53:10:15","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}
23:34:31 SER: Received AT+STATUS?
23:34:31 SER: Transmit AT+STATUS=4

Before the FW upgrade, with Tasmota 6.1.1, everything worked fine.

Any advice on how to fix this?
Do I need to open up the SC and proceed with serial upgrade? (with Atmega chip disconnected)?

tlc76, 2019-07-14:
I finally got back home and I powered off the SC for 10-15 seconds then turned it back on.
and.., problem fixed! The ESP and Atmega are finally communicating correctly:

`3:44:34 SRC: MQTT
23:44:34 RSL: Received Topic tasmota/sonoff-sc/cmnd/POWER, Data Size 3, Data @��
23:44:34 RSL: Group 0, Index 1, Command POWER, Data @��
23:44:34 MQT: tasmota/sonoff-sc/stat/RESULT = {"Command":"Unknown"}
23:44:34 MQT: Subscribe to cmnd/DVES_689F9B_fb/#
23:44:34 MQT: tasmota/sonoff-sc/tele/INFO1 = {"Module":"Sonoff SC","Version":"6.5.0(sonoff)","FallbackTopic":"cmnd/DVES_689F9B_fb/","GroupTopic":"sonoffs"}
23:44:34 MQT: tasmota/sonoff-sc/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-sc","IPAddress":"10.0.0.60"}
23:44:34 MQT: tasmota/sonoff-sc/tele/INFO3 = {"RestartReason":"Power on"}
23:44:34 SRC: MQTT
23:44:34 RSL: Received Topic tasmota/sonoffs/cmnd/status, Data Size 1, Data 
23:44:34 RSL: Group 1, Index 1, Command STATUS, Data 
23:44:34 MQT: tasmota/sonoff-sc/stat/STATUS = {"Status":{"Module":21,"FriendlyName":["Sonoff SC"],"Topic":"sonoff-sc","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:44:35 SER: Received AT+UPDATE="humidity":30,"temperature":27,"light":10,"noise":4,"dusty":1
23:44:35 SER: Transmit AT+SEND=ok
23:44:37 APP: Boot Count 6
23:44:37 CFG: Saved to flash at FA, Count 18, Bytes 3584
23:44:38 SER: Received AT+UPDATE="humidity":38,"temperature":33,"light":10,"noise":4,"dusty":1
23:44:38 SER: Transmit AT+SEND=ok
23:44:42 SER: Received AT+UPDATE="humidity":39,"temperature":34,"light":10,"noise":4,"dusty":1
23:44:42 SER: Transmit AT+SEND=ok
23:44:43 MQT: tasmota/sonoff-sc/tele/STATE = {"Time":"2019-07-14T23:44:43","Uptime":"0T00:00:16","Vcc":3.182,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"dasdadasd","BSSId":"78:11:DC:53:22:22","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}
23:44:43 MQT: tasmota/sonoff-sc/tele/SENSOR = {"Time":"2019-07-14T23:44:43","SonoffSC":{"Temperature":34.0,"Humidity":39.0,"Light":10,"Noise":40,"AirQuality":100},"TempUnit":"C"}`