arendst/Tasmota #1176: Sonoff Dual - On Board Button Headers - need HELP!!

chietanjain, 2017-11-13:
I need help!

Any information on using the Button 0 and Button 1 headers to toggle relay 1 and relay 2 respectively would help.

The headers don't work as expected (toggle relay 1 on shorting button 1&ground and toggle 2 on shorting button 2&ground) when configured as "Sonoff Dual".

I want connect a manual push button switch to the Sonoff while still using it with MQTT. I am unable to as the button headers don't work as they do with stock firmware.

The hardware button on the Sonoff Dual works with single and double click for each of the relays' respectively but I would like to use the headers and use discrete physical push buttons for each relay.

Ive tried every possible setting, but ive been unsuccessful. 

Appreciate any help I can get!!! Thank you!

davidelang, 2017-11-13:
The on board button headers are not supported by tasmota, you can't use them.

chietanjain, 2017-11-13:
@davidelang thank you for your reply.

Is there anyway to use the Daul along with a wall switch. As the Tasmota does not support the onboard headers, is there any other pins I can use to connect a toggle push button for each relay separately? Using the hardware button for single press and double press confuses the rest of my family and they end up pressing it for longer, putting the sonoff into other modes and even sometimes resetting it when its pressed more than 10 seconds.

Thanks!

davidelang, 2017-11-13:
look on the wiki, but I don't think there are any pins that are easy to get at 
on the dual

chietanjain, 2017-11-14:
Thanks.

any idea is these settings are effective for the sonoff dual:

SwitchMode<x> |   | Show current external switch mode
SwitchMode<x> | 0 | (default) Set switch mode to TOGGLE
SwitchMode<x> | 1 | Set switch mode to FOLLOW (0 = Off, 1 = On)
SwitchMode<x> | 2 | Set switch mode to inverted FOLLOW (0 = On, 1 = Off)
SwitchMode<x> | 3 | Set switch mode to PUSHBUTTON (Normally 1, 0 = toggle)
SwitchMode<x> | 4 | Set switch mode to inverted PUSHBUTTON (Normally 0, 1 = toggle)
SwitchMode<x> | 5 | Set switch mode to PUSHBUTTON (Normally 1, 0 = toggle, Hold = hold)
SwitchMode<x> | 6 | Set switch mode to inverted PUSHBUTTON (Normally 0, 1 = toggle, Hold = hold)

Setting SwitchMode to anything doesn't change the default behaviour, so I am asking. Am I missing something?

davidelang, 2017-11-14:
switchmode only affects external switches connected to the ESP GPIO pins, it has 
nothing to do with the sonoff dual button headers because they are connected to 
a different chip (and we don't support that function of that chip)

enesbcs, 2017-12-02:
@chietanjain with ESPEasy plugin P165 you can see if the onboard Button0 and Button1 changes, they are reporting back it's status in serial line, that is handled by the plugin, and can be used with Rules.
https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3245&start=40#p16998

MarioDBianchi, 2018-02-27:
Hi, to have both headers buttons of the Sonoff Dual working properly and as expected, you need to put a pull up resistor between each button pin and VCC, then giving a GND pulse will work as expected. A 10K resistor would be OK.

Hope this helps !

haidavila, 2018-03-08:
Hi MarioDBianchi, can you show us and schematic or simple photo?

Regards!