arendst/Tasmota #1177: Voltage missing in Domoticz config page

SimoneBnc, 2017-11-13:
I was previously running version 5.5.2b on a POW then upgraded to latest stable version and I noticed I no longer have the voltage field in Domoticz page that I had before and it was sending voltage data and it stopped sending voltage readings right after I did the update. Now it will not let me upload OTA a recompiled version of the 5.5.2b. Is there a fix to have the voltage back?

arendst, 2017-11-13:
Mine still works as expected.

Enable weblog 3 to see if pow still sends data to domoticz.

SimoneBnc, 2017-11-13:
It only sends data for IDX of the POWER sensor; in domoticz page I only see up to IDX 6 and I remember I had 2 more if I am not mistaken

arendst, 2017-11-13:
So to solve just enable the idexes again on the Domoticz configuration page

SimoneBnc, 2017-11-13:
[image]
let's hope this makes it more clear

arendst, 2017-11-13:
So you are using a version which does NOT support voltage and current. 

Domoticz Voltage and Current was introduced in development release version 5.8.0b and official release version 5.9.0. See file _releasenotes.ino

[image]

SimoneBnc, 2017-11-13:
I have this
 #define VERSION                0x05090101   // 5.9.1a
Could it be something corrupted on the internal flash?

arendst, 2017-11-13:
What is the result of command status 2?

SimoneBnc, 2017-11-13:
After some tinkering I figured it had an old version on it, don't understand why since I was compiling the 5.9.1. Thanks for your help.

closed 2017-11-13 by SimoneBnc