arendst/Tasmota #1178: tasmota mqtt + PWA

MariusGrey, 2017-11-13:
Hi,
I installed tasmota on a sonoff 4ch pro. It works via HTTP requests, but I cant control it from mqtt (mqttlens + cloudmqtt.com)... do you know why?
I have a website (wordpress) and I use ita from smartphone (mobile theme + progressive web app tool), now I'd like to use this PWA to controll my sonoff 4ch pro. I cant use HTTP requests because PWA need an SSL connection, so I think to use MQTT via PHP... is it possible? Do you know how? Can you help me?
THX

davidelang, 2017-11-13:
you need to start out by figuring out if you are connecting to your mqtt server 
and passing messages (you should be able to see this in the logs)

from there, you need to figure out how to send the appropriate messages from 
whatever you connect to your mqtt server.

you haven't given us any detail about what fails in your message, so I can't 
give you any details of how to solve it, just where to start looking.

MariusGrey, 2017-11-13:
Hi davidelang,
thanks for your reply.
I just put the sonoff online.
Here some SS:
[image]
[image]
[image]
[image]

Can you send command to it?
Can you check it?

THX

MariusGrey, 2017-11-14:
Is still online. Do you need more info?

MariusGrey, 2017-11-14:
help me, please. 
Do you need more details?

closed 2018-05-07 by stale[bot]