arendst/Tasmota #1179: Sonoff Dual setting module as "Dual" does not switch relays

peterforeman, 2017-11-13:
I've just installed a new Sonoff Dual as replacement for a Sonoff Basic. Now the fun part is: when configuring it as "dual" neither relays work. When configuring it as "Basic" (which it isn't) makes me switch Relay1 via the "Toggle" button on the webinterface.

Is this a new type of dual which does not use the serial switching logic anymore?

chietanjain, 2017-11-14:
Are you using MQTT?

peterforeman, 2017-11-14:
Yes, I'm using MQTT, but also that does not work when using POWER1/POWER2.

flotre, 2017-11-20:
I have same issue.
Sonoff dual 5.9.1

Physical Relay don't work using mqtt, web interface, console.

And button don't work too.

Tresett, 2017-11-21:
Are you disable serial port?

21 нояб. 2017 г. 2:24 AM пользователь "flotre" <notifications@github.com>
написал:

I have same issue.
Sonoff dual 5.9.1

Physical Relay don't work using mqtt, web interface, console.

And button don't work too.

flotre, 2017-11-21:
No, just configure firmware to sonoff dual

When configure as sonoff 4ch, it works.

arendst, 2017-11-21:
Pls provide pictures and a link from where you bought it. This sounds to me like totally different hardware than the original dual using the serial interface to control two relays.

saundersrg, 2017-12-13:
Having the same thing my side, LED changes, but relays don't kick over. Sonoff Dual Ver 2.0. SerialLog set to 0.

Chips to right of ESP8266EX are as follows from what I can see.

XTX
BN25F08
1(?)644Q0R

SIL
F330
B0GAZ
1624+

[image]
[image]
[image]
[image]

FuzzzzzyLogic, 2018-01-04:
Upgrading Tasmota from 5.9 to 5.10 fixed this for me, since it adds SONOFF_DUAL_R2.

- If updating in user_config.h, after flashing you may need to use the web interface to do `Console->"Reset 2"` to clear any stored override.
- Otherwise, you can update this in the web interface with `Configuration->Configure Module` and select `Module Type "39 Sonoff Dual R2"`.

peterforeman, 2018-03-09:
Problem is solved for me using the 39 Sonoff Dual R2, closing.

closed 2018-03-09 by peterforeman