arendst/Tasmota #1180: Sonoff not working with low power LED lights

mattpitts74, 2017-11-13:
Hi There, 

I have been struggling to get my sonoff working with a small low lower LED desk lamp.  I was able to flash OK and tested connected via the header pins, so I know the sonoff is working and the replay switches.  But when I connect it to the actual lamp nothing happens.  The sonoff doesn't seem to power up and never appears on my network. 

 I have tried swapping the connections around and reconnecting the whole thing. I’ve even completely removed the Sonoff and plugged the lamp directly back into the mains just to check its working. I’ve also tried several other Sonff’s just to check it wasn’t a faulty one. The only thing I haven’t checked in with a multi meter as I don’t have one.

smadds, 2017-11-13:
Have you joined it to your Wifi network? Press the button 4 times soon after power up to put it into access point mode, then look for it as a new wifi network using your phone for instance. Connect to that Wifi on your phone, then open a browser on it pointing to 192.168.4.1 to configure the Sonoff

mattpitts74, 2017-11-13:
Thanks, so far I haven't had to do what you suggested with the other sonoff's that I have connected to my network, how important is the positive and negative wires, does it matter if they are connected the wrong way around?  my light has a moulded plug which is impossible to take apart!

smadds, 2017-11-13:
If you mean the colours of wires in the mains cable (normally brown & blue), it is not important. If it's a 2-core cable it's easy to fit a basic Sonoff inline, somewhere along the cable. If it's a 3core then a POW is easier to connect.

mattpitts74, 2017-11-13:
I still cannot get this working, I've just been checking the plug power adapter of the LED lights I've connected inline to and its says Output 12 volts 2.01 amps  https://photos.app.goo.gl/1wozXWOeDXzXuGL73

smadds, 2017-11-13:
Aha - the mist is clearing...

I think you have 2 options, @mattpitts74 

1. For just on/off use an S20 socket to plug the LEDs into
2. For dimming, get a Tasmota compatible low-voltage dimmer. I use something called a H801 (search ebay). It has 5 channels of dimming - you may only need one of them.

the normal Sonoffs will not work on the 12V low voltage as they are expecting mains voltage to power them.

mattpitts74, 2017-11-13:
Thanks its good to have that confirmed, I have been pulling my hair out with this thing most of the day, but glad to have figured out why it wasn't working, should have checked the power before hand!  Thanks for your help figuring this out.

qtronik, 2017-12-02:
Se can now close this tread...

closed 2018-05-07 by stale[bot]