arendst/Tasmota #1181: RF Bridge Saving Codes... struggling

ins4neuk, 2017-11-13:
Can any one help me with saving codes on the rf bridge please i'm struggling with getting them to save.

I can get the device to see the codes when i press the button but i have no idea on assigning it to a key. every time i press a key or do one of the rfkey commands i just always get a console line "stat/sonoff/RESULT = {"RfKey1":"Learning active"}"

zibous, 2017-11-17:
Same problem here, can not put the RF Bridge in learning mode.

```
14:53:18 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9280, "Low":310, "High":880, "Data":"000551", "RfKey":"None"}}
14:53:19 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9330, "Low":300, "High":900, "Data":"00155F", "RfKey":"None"}}

```

When typing in the console I get the message:
```
14:59:30 CMD: RfKey1:2
14:59:30 MQT: stat/RFB119/RESULT = {"Command":"Unknown"}

```

Program Version | 5.9.0a
2_3_0/1.5.3(aec24ac9)

arendst, 2017-11-17:
Where does the : come from? Try ``rfkey 2``.

zibous, 2017-11-17:
Now i can go to learning mode (input from console)
```
 RfKey1 2
 rfkey 2
```
Works too, signal is received and confirmed with a beep. 
But then i can never stopped learning mode or switch to another RfKey for
new learning...

```
15:36:40 CMD: rfkey 2
15:36:40 MQT: stat/RFB119/RESULT = {"RfKey1":"Learning active"}

15:29:24 CMD:  RfKey1 4
15:29:24 MQT: stat/RFB119/RESULT = {"RfKey1":"Learning active"}
15:29:36 CMD:  RfKey1
15:29:36 MQT: stat/RFB119/RESULT = {"RfKey1":"Learning active"}
```
New Version deployed:  
Program Version | 5.9.1d
-- | --

arendst, 2017-11-17:
Every time you select option 2 it goes in learning mode waiting for a key press or times out after a minute.

rfkey is equal to rfkey1. 

As the bridge is still in learning mode (Learning active) you cannot select other commands.

zibous, 2017-11-17:
Thanks for this information, but the learning mode never ends, only when i restart the device..
```
15:46:20 CMD: rfkey
15:46:20 MQT: stat/RFB119/RESULT = {"RfKey1":"Learning active"}
15:46:24 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9270,"Low":310,"High":880,"Data":"00155F","RfKey":"None"}}
15:46:26 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9270,"Low":310,"High":880,"Data":"00055F","RfKey":"None"}}
15:46:27 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9280,"Low":310,"High":880,"Data":"00155F","RfKey":"None"}}
15:46:28 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9280,"Low":310,"High":880,"Data":"000551","RfKey":"None"}}
15:46:28 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9280,"Low":310,"High":880,"Data":"00155F","RfKey":"None"}}
15:46:31 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9270,"Low":310,"High":880,"Data":"00155F","RfKey":"None"}}
15:46:34 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9290,"Low":310,"High":880,"Data":"000551","RfKey":"None"}}
15:46:35 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9280,"Low":310,"High":880,"Data":"00155F","RfKey":"None"}}
15:46:42 MQT: tele/RFB119/STATE = {"Time":"2017-11-17T15:46:42","Uptime":0,"Vcc":3.218,"Wifi":{"AP":1,"SSId":"McWille","RSSI":100,"APMac":"C4:B3:01:DC:55:B6"}}
15:46:49 CMD: rfkey
15:46:49 MQT: stat/RFB119/RESULT = {"RfKey1":"Learning active"}
---
15:49:39 MQT: tele/RFB119/INFO3 = {"RestartReason":"Software/System restart"}
15:49:47 MQT: tele/RFB119/STATE = {"Time":"2017-11-17T15:49:47","Uptime":0,"Vcc":3.217,"Wifi":{"AP":1,"SSId":"McWille","RSSI":100,"APMac":"C4:B3:01:DC:55:B6"}}
15:49:52 CMD: rfkey
15:49:52 MQT: stat/RFB119/RESULT = {"RfKey1":"Default sent"}
```

arendst, 2017-11-17:
This should have happened:
```
15:53:05 HTP: Main Menu
15:53:08 HTP: Console
15:53:12 CMD: rfkey1
15:53:12 RSL: Group 0, Index 1, Command RFKEY, Data 
15:53:12 MQT: stat/bridge1/RFKEY = {"RfKey1":"Default sent"}
15:53:20 CMD: rfkey1 2
15:53:20 RSL: Group 0, Index 1, Command RFKEY, Data 2
15:53:20 MQT: stat/bridge1/RFKEY = {"RfKey1":"Start learning"}
15:54:20 BRG: Received A2 55 
15:54:20 MQT: stat/bridge1/RFKEY = {"RfKey1":"Learn failed"}
15:54:37 CMD: rfkey1 2
15:54:37 RSL: Group 0, Index 1, Command RFKEY, Data 2
15:54:37 MQT: stat/bridge1/RFKEY = {"RfKey1":"Start learning"}
15:54:41 BRG: Received A3 20 DA 01 18 03 34 2E 1A 22 55 
15:54:41 MQT: stat/bridge1/RFKEY = {"RfKey1":"Learned"}
15:54:43 CFG: Saved to flash at F9, Count 476, Bytes 1648
15:54:56 CMD: rfkey1
15:54:56 RSL: Group 0, Index 1, Command RFKEY, Data 
15:54:56 MQT: stat/bridge1/RFKEY = {"RfKey1":"Learned sent"}
```

zibous, 2017-11-17:
Hi,
Negativ, can not see this messages..

```
15:58:41 CMD: rfkey 2
(Beep)
15:58:41 MQT: stat/RFB119/RESULT = {"RfKey1":"Start learning"}
(Beep, Beep)
```
Learning will never stop..

arendst, 2017-11-17:
- Make sure seriallog = 0
- Enable weblog 3 to see some serial communication between esp8266 and RF chip

zibous, 2017-11-17:
No i can see more, but strange messages

```
16:16:05 CMD: seriallog = 0
16:16:05 MQT: stat/RFB119/RESULT = {"SerialLog":"4 (Active 4)"}
16:16:37 CMD: weblog 3
16:16:37 MQT: stat/RFB119/RESULT = {"WebLog":3}
16:16:37 CFG: Saved to flash at F5, Count 24, Bytes 1648
16:18:04 CMD: rfkey
16:18:04 RSL: Group 0, Index 1, Command RFKEY, Data 
16:18:04 MQT: stat/RFB119/RESULT = {"RfKey1":"Learning active"}
16:18:09 BRG: Received A4 24 68 01 36 03 7A 00 15 5F 55 
16:18:09 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9320,"Low":310,"High":890,"Data":"00155F","RfKey":"None"}}
16:19:15 MQT: tele/RFB119/STATE = {"Time":"2017-11-17T16:19:15","Uptime":0,"Vcc":3.218,"Wifi":{"AP":1,"SSId":"McWille","RSSI":100,"APMac":"C4:B3:01:DC:55:B6"}}
16:19:17 RSL: Group 0, Index 1, Command RFKEY, Data 
16:19:17 MQT: stat/RFB119/RESULT = {"RfKey1":"Learning active"}
16:19:18 RSL: Group 0, Index 2, Command RFKEY, Data 
16:19:18 MQT: stat/RFB119/RESULT = {"RfKey1":"Learning active"}
16:19:21 BRG: Received A4 24 54 01 36 03 7A 00 15 5F 55 
16:19:21 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9300,"Low":310,"High":890,"Data":"00155F","RfKey":"None"}}
16:19:22 RSL: Group 0, Index 1, Command RFKEY, Data 
16:19:23 MQT: stat/RFB119/RESULT = {"RfKey1":"Learning active"}
-- Back to main menu and than to console...
16:19:48 HTP: Console
16:19:51 MQT: Attempting connection...
16:19:51 MQT: Connected
16:19:51 MQT: tele/RFB119/LWT = Online (retained)
16:19:51 MQT: cmnd/RFB119/POWER = 
16:19:51 MQT: Subscribe to cmnd/RFB119/#
16:19:51 MQT: Subscribe to cmnd/smarthome/#
16:19:51 MQT: Subscribe to cmnd/DVES_8F42C0/#
16:20:01 CMD: rfkey1
16:20:01 RSL: Group 0, Index 1, Command RFKEY, Data 
16:20:01 MQT: stat/RFB119/RESULT = {"RfKey1":"Learning active"}
16:20:19 BRG: Received A4 15 EA 01 FE 03 F2 EB 30 BC 55 
16:20:19 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":5610,"Low":510,"High":1010,"Data":"EB30BC","RfKey":"None"}}
```

arendst, 2017-11-17:
As your seriallog was set to 4 the RF chip was overcrowded with rubish data. No restart one more time and try again.

zibous, 2017-11-17:
Restarted but nothing has changed...

```
16:27:42 MQT: Subscribe to cmnd/DVES_8F42C0/#
16:27:42 MQT: tele/RFB119/INFO1 = {"Module":"Sonoff Bridge","Version":"5.9.1d","FallbackTopic":"DVES_8F42C0","GroupTopic":"smarthome"}
16:27:42 MQT: tele/RFB119/INFO2 = {"WebServerMode":"Admin","Hostname":"IPS-RFB119","IPAddress":"10.1.1.119"}
16:27:42 MQT: tele/RFB119/INFO3 = {"RestartReason":"Software/System restart"}
16:27:43 HTP: Main Menu
16:27:45 HTP: Information
16:27:50 MQT: tele/RFB119/STATE = {"Time":"2017-11-17T16:27:50","Uptime":0,"Vcc":3.218,"Wifi":{"AP":1,"SSId":"McWille","RSSI":100,"APMac":"C4:B3:01:DC:55:B6"}}
16:27:52 HTP: Information
16:27:55 HTP: Console
16:28:00 CMD: rfkey1
16:28:00 RSL: Group 0, Index 1, Command RFKEY, Data 
16:28:01 MQT: stat/RFB119/RESULT = {"RfKey1":"Default sent"}
16:28:34 CMD: rfkey1 2
16:28:34 RSL: Group 0, Index 1, Command RFKEY, Data 2
16:28:34 MQT: stat/RFB119/RESULT = {"RfKey1":"Start learning"}
16:28:49 CMD: rfkey1
16:28:49 RSL: Group 0, Index 1, Command RFKEY, Data 
16:28:49 MQT: stat/RFB119/RESULT = {"RfKey1":"Learning active"}
16:28:50 MQT: tele/RFB119/STATE = {"Time":"2017-11-17T16:28:50","Uptime":0,"Vcc":3.218,"Wifi":{"AP":1,"SSId":"McWille","RSSI":100,"APMac":"C4:B3:01:DC:55:B6"}}
16:29:24 HTP: Main Menu
16:29:27 HTP: Console
16:29:34 BRG: Received A4 24 68 01 36 03 7A 00 15 5F 55 
16:29:34 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9320,"Low":310,"High":890,"Data":"00155F","RfKey":"None"}}
16:29:35 CMD: rfkey
16:29:35 RSL: Group 0, Index 1, Command RFKEY, Data 
16:29:35 MQT: stat/RFB119/RESULT = {"RfKey1":"Learning active"}
16:29:50 MQT: tele/RFB119/STATE = {"Time":"2017-11-17T16:29:50","Uptime":0,"Vcc":3.218,"Wifi":{"AP":1,"SSId":"McWille","RSSI":100,"APMac":"C4:B3:01:DC:55:B6"}}
16:30:07 CMD: Status
16:30:07 RSL: Group 0, Index 1, Command STATUS, Data 
16:30:08 MQT: stat/RFB119/STATUS = {"Status":{"Module":25,"FriendlyName":"Smarthome","Topic":"RFB119","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
```

arendst, 2017-11-17:
At 16:28:34 Tasmota is waiting for you to press a key on your physical remote control to be learned by Tasmota.

At 16:29:34 Tasmota receives a signal back from the rf chip apparently containing the code to learn. The rf chip should have replied immediatly after you pressed the remote control button and also signaled the end of learn mode.

It looks to me the software of the rf chip has changed and does not really posses a learn mode anymore.

As a side track I managed to get the same kind of error as you did. Apparently the RF chip does not recognize a received signal and fails to send the end of learn code after 1 minute but does send the unrecognized signal. As tasmota is still waiting for the end of learn mode signal it seems to hang.

I just made a fix which will let tasmota timeout after 60 seconds if no end of learn mode is received from the rf chip. This will solve your endless Learning Active message but still won't recognize your remote control code.

From your logging I may have to rewrite the learn process completely...

zibous, 2017-11-17:
Hi,
```
It looks to me the software of the rf chip has changed and does not really posses a learn mode anymore.
```
Could be, because the device is brand new and has the new board.

Thanks for your help.

ins4neuk, 2017-11-18:
mines still doing the same, beeps and then the red light flashes and just hangs

1:39:39 CMD: RfKey1 2
01:39:39 RSL: Group 0, Index 1, Command RFKEY, Data 2
01:39:39 MQT: stat/sonoff/RESULT = {"RfKey1":"Start learning"}
01:40:15 HTP: Information
01:40:39 BRG: Received A4 34 BC 01 B8 05 14 15 55 
01:40:39 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":13500,"Low":440,"High":1300,"Data":"155500","RfKey":"None"}}

i'll try and upgrade the software to the newer version and see what that does

zibous, 2017-11-18:
Hi Theo,

Is there any way to register the RFKeys with a MQTT command ?
```
cmnd/DVES_8F42C0/RfKey1  {"Command":{"Bit":24,"Protocol":1,"Data":"1364","RfKey":"1"}
cmnd/DVES_8F42C0/data   {"value":"1365","raw":"9050,2300,550","bits":24,"protocol":1,"RfKey":"1"}
```

I also do not know how I can execute a command with MQTT:
```
cmnd/DVES_8F42C0/RfKey1 ON 
cmnd/DVES_8F42C0/RfKey1 OFF
```

More about "Sonoff 433 RF Bridge won't pair" problem:
https://itead.freshdesk.com/support/discussions/topics/11000014948
https://itead.freshdesk.com/support/discussions/topics/11000015088
https://www.youtube.com/watch?v=luRHzcc5f8Q&feature=youtu.be

Thanks

binderth, 2017-11-19:
Found time for my (also new platine) Sonoff RF Bridge 433 - I am experiencing the same thing (v5.9.1e):

1. fresh started device
2. open `console`
3. `RfKey1 2`
4. `11:06:08 CMD: RfKey1 2`
`11:06:09 MQT: stat/sonoff/RESULT = {"RfKey2":"Start learning"}`
4. Sonoff RF Bridge 433 beeps
5. *pressing the 433button*
6. Sonoff RF Bridge 433 quits/beeps
_repeat step 6 multiple times_
7. `RfKey1 4`
8. `11:06:33 CMD: RfKey2 4`
`11:06:34 MQT: stat/sonoff/RESULT = {"RfKey1":"Learning active"}`

Is there something I can do to provide more info?

arendst, 2017-11-19:
@zibous You can save values to rfkeys using the individual commands like:
```
backlog rfsync 8410; rflow 280; rfhigh 830; rfcode #2e1a22; rfkey2 4
```
The above sequence of commands saves it's code for rfkey2

@binderth First get to know the functions of the option numbers behind the rfkey<x> command as documented in the wiki:
```
RfKey<x> |   | Send learned or default RF data for RfKey1 to RfKey16
RfKey<x> | 1 | Send default RF data for RfKey1 to RfKey16 using RfSync, RfLow, RfHigh and RfHost parameters
RfKey<x> | 2 | Learn RF data for RfKey1 to RfKey16
RfKey<x> | 3 | Unlearn RF data for RfKey1 to RfKey16
RfKey<x> | 4 | Save RF data using RfSync, RfLow, RfHigh and last RfCode parameters
```See above answer regarding use of option 4.

For debugging purposes it is good to know if your remote control is able to control your 433MHz device directly; Not all remotes are recognised by the Sonoff Bridge RF hardware. Tasmota is NOT decoding the remote control codes but a seperate chip inside the bridge is doing this and if it cannot decode the code it just fails to send the valid learn status to Tasmota.

binderth, 2017-11-19:
@arendst: thanks for the heads up! Now after some tries I could learn a remote button to RfKey4:
```
13:51:14 CMD: RfKey4 2
13:51:14 MQT: stat/sonoff/RESULT = {"RfKey4":"Start learning"}
13:51:32 MQT: stat/sonoff/RESULT = {"RfKey4":"Learned"}
13:51:34 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":5700, "Low":190, "High":540, "Data":"507502", "RfKey":"None"}}
13:51:43 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":5680, "Low":200, "High":530, "Data":"505502", "RfKey":"None"}}
13:51:52 CMD: RfKey4
13:51:52 MQT: stat/sonoff/RESULT = {"RfKey4":"Learned sent"}
```
again: i pressed the button 4times, so I guess you have to press it at least two times for the learning. After learning `RfKey4` now triggers my RF outlet. Great Job!

zibous, 2017-11-23:
Hi Theo,

Now with Program Version | 5.9.1h i can learn RfKey1 and this will work.
But when i can't learn RfKey2, RfKey3.

RfKey1 2  works
RfKey2 2 not learning failed

arendst, 2017-11-24:
No problem at all:
```
11:17:04 CMD: rfkey2 2
11:17:05 RSL: Group 0, Index 2, Command RFKEY, Data 2
11:17:05 MQT: stat/bridge1/RFKEY = {"RfKey2":"Start learning"}
11:17:09 BRG: Received A3 20 E4 01 18 03 3E 2E 1A 28 55 
11:17:10 MQT: stat/bridge1/RFKEY = {"RfKey2":"Learned"}
11:17:10 CFG: Saved to flash at F9, Count 499, Bytes 1648
```

zibous, 2017-11-24:
Try to register rfkey1 and rfkey2:

```
rfkey1 2
11:42:50 CMD: rfkey1 2
11:42:50 RSL: Received Topic /rfkey1, Data Size 1, Data 2
11:42:50 RSL: Group 0, Index 1, Command RFKEY, Data 2
-- Beep
11:42:50 MQT: stat/RFB199/RESULT = {"RfKey1":"Start learning"}
-- Press Key
-- Beep Beep
11:43:50 BRG: Received A4 24 4A 01 36 03 70 00 15 5F 55 
11:43:50 MQT: stat/RFB199/RESULT = {"RfKey1":"Learn failed"}
11:45:02 CMD: rfkey1
11:45:02 RSL: Received Topic /rfkey1, Data Size 0, Data 
11:45:02 RSL: Group 0, Index 1, Command RFKEY, Data 
11:45:02 MQT: stat/RFB199/RESULT = {"RfKey1":"Learned sent"}
-- Switch ON (works)
11:46:34 CMD: rfkey2 2
-- Beep
11:46:34 RSL: Received Topic /rfkey2, Data Size 1, Data 2
11:46:34 RSL: Group 0, Index 2, Command RFKEY, Data 2
11:46:34 MQT: stat/RFB199/RESULT = {"RfKey2":"Start learning"}
-- Press Key
-- Beep Beep
11:47:34 BRG: Received A4 24 90 01 2C 03 7A 00 05 5F 55 
11:47:34 MQT: stat/RFB199/RESULT = {"RfKey2":"Learn failed"}
----
11:49:34 CMD: rfkey1
11:49:34 RSL: Received Topic /rfkey1, Data Size 0, Data 
11:49:34 RSL: Group 0, Index 1, Command RFKEY, Data 
11:49:34 MQT: stat/RFB199/RESULT = {"RfKey1":"Learned sent"}
-- Switch ON (works)
11:50:05 CMD: rfkey2
11:50:05 RSL: Received Topic /rfkey2, Data Size 0, Data 
11:50:05 RSL: Group 0, Index 2, Command RFKEY, Data 
11:50:05 MQT: stat/RFB199/RESULT = {"RfKey2":"Default sent"}
-- Nothing happens..

Press Key1 on Remote Control
11:51:34 BRG: Received A4 24 68 01 2C 03 7A 00 15 5F 55 
11:51:34 MQT: tele/RFB199/RESULT = {"RfReceived":{"Sync":9320,"Low":300,"High":890,"Data":"00155F","RfKey":"None"}}

Press Key2 on Remote Control
11:52:09 BRG: Received A4 24 72 01 2C 03 7A 00 05 5F 55 
11:52:09 MQT: tele/RFB199/RESULT = {"RfReceived":{"Sync":9330,"Low":300,"High":890,"Data":"00055F","RfKey":2}}
```

Unfortunately does not work with the backlog command:
```
backlog rfsync 9310; rflow 310 rfhigh 890; rfcode #00155F; rfkey1 4
backlog rfsync 9300; rflow 310 rfhigh 890; rfcode #00055F; rfkey2 4
```

When I use the remote control and control this with an ESP8266 (RCSwitch Lib) then
i can use all codes to switch the devices.

```
{"device":"HFYJ18","mode":"RF-RX","protocol":1,"bits":24,"value":1361,"repeat":1}
{"device":"HFYJ18","mode":"RF-RX","protocol":1,"bits":24,"value":1364,"repeat":1}
```

Anpro 4x 433 MHz Empfänger und Funk Sende Modul Superregeneration 
Wireless Transmitter Modul Einbrecher Alarm Burglar Alarm mit 2 Helical Antenne
https://www.amazon.de/gp/product/B071J2Z3YK/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1

Modellbezeichnung: Arendo - Funksteckdosen-Set (4+1) für den Außenbereich (Outdoor)
https://www.amazon.de/gp/product/B00QXCRI2M/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

It's a pity, because I really like the Sonoff device.
Thanks.

tatunts, 2017-12-03:
Hello everyone!

Sorry to bother you: I just received my RF Bridge from ITEAD and successfully flashed with TASMOTA latest version (5.10).

I was unable to learn any of my 433 codes that I currently use with my Broadlink RM PRO. It is very strange, my device does not beep when command `rfkey1 2` is issued on console.

It hangs like this:

```
14:51:18 CMD: RfKey1 2
14:51:18 MQT: stat/rfbridgecasa/RESULT = {"RfKey1":"Start learning"}
```

There is no beep and nothing happens when I press the controller button.

I've noticed that the board does not seems to be the same as the ones used on the Wiki pages. Here it is: https://photos.app.goo.gl/Vn7Di5bsDRSvOdKu2

Can someone help me?

Thanks!

Wuckel01, 2017-12-30:
Hello

I have rf bridge and wanted to use the door contact magnet to receive the status on openhab.
how do I register the contact to use status.
maybe someone can help me?
Thanks

jaqPi, 2018-01-01:
Hello there!

My Sonoff also has problems to learn the code of my 433 MHz remote.

**My settings**
- The _new_ Sonoff (like the one of @tatunts)
- Firmware version 5.10.0g
- Brennenstuhl RCS 1000 N Comfort [https://www.amazon.de/Brennenstuhl-Funkschalt-Set-Funksteckdosen-Innenbereich-Kindersicherung/dp/B001AX8QUM](url)

**Log output**
22:13:54 CMD: weblog = 3
22:13:54 RSL: Group 0, Index 1, Command WEBLOG, Data = 3
22:13:54 RSL: stat/sonoff/RESULT = {"WebLog":3}
22:14:56 CMD: seriallog 0
22:14:56 RSL: Group 0, Index 1, Command SERIALLOG, Data 0
22:14:56 RSL: stat/sonoff/RESULT = {"SerialLog":"0 (Active 0)"}
22:22:12 CMD: rfkey1 2
22:22:12 RSL: Group 0, Index 1, Command RFKEY, Data 2
22:22:12 RSL: stat/sonoff/RESULT = {"RfKey1":"Start learning"}
22:22:12 BRG: Received A0 55 
22:23:12 BRG: Received A2 55 
22:23:12 RSL: stat/sonoff/RESULT = {"RfKey1":"Learn failed"}

ciotlosm, 2018-02-09:
For those that have problems learning codes with Sonoff RF Bridge because of "No beep" and "Timeout" there seems to be a problem with the firmware on the EFM8 chip. You will need to upgrade the firmware using something like: https://github.com/rhx/RF-Bridge-EFM8BB1

ornolio, 2018-04-19:
Sorry, to bother everyone, but I would like to add my contribute to this issue: a DW1 not working with a Tasmota 12.0 (upgraded last bin) flashed RF Bridge 433, but it does work OK added with ewlink to a brand new RF BRIDGE 433 running the original ITEAD firmware.

Here under is the story:
I've the same problem discussed on this thread with one out  of 2xDW1 running on an "old case" RF Bridge 433; while second DW1, PIR and remote 433 commands are learned and work ok on it. 

Here is the log of trying to learn the not learned DW1 (let's call it NLDW1):
09:43:35 MQT: tele/bridge-1/STATE = {"Time":"2018-04-19T09:43:35","Uptime":"0T00:00:13","Vcc":3.182,"Wifi":{"AP":1,"SSId":"TIM-24084959","RSSI":100,"APMac":"10:13:31:70:95:F3"}}
09:44:15 CMD: rfkey7 2
09:44:15 RSL: Group 0, Index 7, Command RFKEY, Data 2
09:44:15 MQT: stat/bridge-1/RESULT = {"RfKey7":"Start learning"}
09:44:15 SER: Received A0 55
09:45:15 SER: Received A4 39 3A 01 EA 05 B4 26 F3 41 55
09:45:15 MQT: stat/bridge-1/RESULT = {"RfKey7":"Learn failed"}
-------------------------------tried to test NLDW1, received rfkey "none" ------------------------
09:47:12 SER: Received A4 38 04 01 F4 05 BE 26 F3 41 55
09:47:12 MQT: tele/bridge-1/RESULT = {"RfReceived":{"Sync":14340,"Low":500,"High":1470,"Data":"26F341","RfKey":"None"}}

------------------- tried to flash NLDW1 using the option 4, but still unsuccesful --------------
09:50:48 CMD: rfkey7 4
09:50:48 RSL: Group 0, Index 7, Command RFKEY, Data 4
09:50:48 MQT: stat/bridge-1/RESULT = {"RfKey7":"Saved"}
09:50:49 CFG: Saved to flash at F9, Count 62, Bytes 1648
09:52:03 SER: Received A4 30 C0 01 9A 04 C4 26 F3 41 55
09:52:03 MQT: tele/bridge-1/RESULT = {"RfReceived":{"Sync":12480,"Low":410,"High":1220,"Data":"26F341","RfKey":"None"}}

Unsuccesfully tried on NLDW1 to replace 12V battery A27 (fully loaded, small suitable for remote commands) to 12V 23AE one (fully loaded higher capacity). 

Succesfully I learned a 433 remote command of the same slot "7" on the same RF Bridge 433 :
09:54:35 CMD: rfkey7 2
09:54:35 RSL: Group 0, Index 7, Command RFKEY, Data 2
09:54:35 MQT: stat/bridge-1/RESULT = {"RfKey7":"Start learning"}
09:54:36 SER: Received A0 55
09:54:38 SER: Received A3 33 E0 01 B8 04 EC FE 77 31 55
09:54:38 MQT: stat/bridge-1/RESULT = {"RfKey7":"Learned"}
09:54:39 CFG: Saved to flash at F8, Count 63, Bytes 1648
09:54:50 SER: Received A4 33 C2 01 B8 04 EC FE 77 31 55
09:54:50 MQT: tele/bridge-1/RESULT = {"RfReceived":{"Sync":13250,"Low":440,"High":1260,"Data":"FE7731","RfKey":7}}

Finally I went to add the NLDW1 to another RF BRIDGE 433 standard ITEAD firmware using ewelink at evething worked perfectly.

Conclusion: a Tasmota flashad RFBRIDGE 433  learns PIR, remote commands and DW1 but NOT this specific NLDW1, which on the contrary is learned by ITEAD firwmware on anothe RF BRIDGE 433

Wuckel01, 2018-04-19:
Hello,
I also had this problem. No pipton in the log, nothing was displayed. My mistake was really stupid, I had not switched back the program switch. there was nothing going on.
Thanks for the really great help.

ornolio, 2018-04-20:
continue my previous comment:
moved the not learning DW1 (NLDW1) to another location and learned immediately into another RF BRIDGE 433 flashed with the same Tasmota rel. 12.0.
hoping this could help.

zeliant, 2018-04-20:
I'm having trouble to learn the RF codes of my remote as well.

I started the learning by entering rfkey1 2 but it doesn't pick up any codes after I repeatedly press my remote.

16:54:50 CMD: rfkey1 2
16:54:50 MQT: stat/sonoffRF/RESULT = {"RfKey1":"Start learning"}
16:55:50 MQT: stat/sonoffRF/RESULT = {"RfKey1":"Learn failed"}

My sonoff bridge is different from the model in the wiki, could this be the reason?

sarsw, 2018-06-23:
zeliant did you solve your learning RF code issue?

I posted here https://github.com/arendst/Sonoff-Tasmota/issues/3025 with issues and now have the "Start Learning" and "Learning Failed" issue.

Jason2866, 2018-06-25:
This code
`14:53:18 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9280, "Low":310, "High":880, "Data":"000551", "RfKey":"None"}}
14:53:19 MQT: tele/RFB119/RESULT = {"RfReceived":{"Sync":9330, "Low":300, "High":900, "Data":"00155F", "RfKey":"None"}}
`
is IT "Intertechno" or Code 2262 and the orig. Firmware of the EFM Chip cant decode this correct.
With Tasmota V.6.0.0a now you can flash a alternative Version (Portisch) and it works!

mc1271, 2018-07-05:
@Jason2866 struggling with my RF Bridge and Pairing Sonoff T433 86 Buttons.
Updated to V6.0.0c but with same negativ result. ("Learn failed")

Any ideas ? thx

Jason2866, 2018-07-06:
@mc1271 
Do you have flashed the alternative EFM Firmware too?

mc1271, 2018-07-06:
No, I’m still with the Orginal Version.
I was under the impression that a SonOff Bridge should be able to talk to a SonOff Switch.
Is there a way to flash OTA via 6.0.0c ? Or do I need to solder and flash again ?

mc1271, 2018-07-06:
As I’m new to RF I might be thinking wrong. All I’m trying to do is assigning the T433 86 as Key 1 in order the use the RFKey in OpenHAB to trigger Events.
I see traffic in the Bridge Console, but RFKey remains empty because learning fails.
Am I on the right Track ?

Jason2866, 2018-07-06:
The keys in the bridge are predefined with working rf codes.
Just press the virtual button 1 in web console and let sonoff device learn this code.

mc1271, 2018-07-06:
Isn’t that the same as entering:
RfKey1 2 and
RfKey1 4 via Console (and Pressing the T 433 86 Switch) several times?

Jason2866, 2018-07-06:
No. Rfkey1 1 will send predefined code!!
Rfkey1 2 starts learning key for button 1
The T433 has to learn the code SENDED from the bridge
Read the wiki for the T switch

mc1271, 2018-07-07:
Hi Jason, there might be a missunderstanding. 
I want to connect a [T433 86 Button](https://www.itead.cc/smart-home/t433-sticky-rf-remote.html)
(That's purely a remote button) to my SonOff RF Bridge for it to send an RFKey towards OpenHab in order to process that in my rules. So currently I get:
`12:41:09 MQT: tele/sonoffbridge/RESULT = {"RfReceived":{"Sync":10030,"Low":320,"High":970,"Data":"568ED2","RfKey":"None"}}`
my desired result would be:
`12:41:09 MQT: tele/sonoffbridge/RESULT = {"RfReceived":{"Sync":10030,"Low":320,"High":970,"Data":"568ED2","RfKey":"1"}}`
 I didn't find anything about the T433 in the WIKI

Jason2866, 2018-07-07:
Yes  i missundrrstood you. You dont have to learn the rfbridge the T433!
I use OpenHab too. Just read the value of Data -> 568ED2 in your case in a switch item definition.
With that you can do all the magic you want with Rules in OpenHab. It is as simple as it sounds.
If you want i can paste a view OpenHab eamples....

Jason2866, 2018-07-07:
Rfbridge.items
String RfBridge_RfReceive "RfReceive1 [%s]"
    { mqtt="<[broker:tele/sonoff-bridge/RESULT:state:JSONPATH($.RfReceived.Data)]" }

Jason2866, 2018-07-07:
Rfbridge.rules
rule "RF433 switch sonoff and Update RF Switch (Bridge 1)"
when
    Item RfBridge_RfReceive changed
then
if (RfBridge_RfReceive.state == NULL)
    {
    logInfo("RfBridge1.rules", "Item is null, cancelling...")
    return;
    } 
    var rfData = RfBridge_RfReceive.state.toString
    logInfo("rfbridge1.rules", "Received IT Codes: " + rfData)
        switch (rfData) {         
            case "A00A1F":
               {
                publish("broker", "cmnd/sonoff-13D92F/POWER", "ON")
                }
            case "A00A17":
                {
                publish("broker", "cmnd/sonoff-13D92F/POWER", "OFF")
                }
            case "A00A1B":
               {
                publish("broker", "cmnd/sonoff-065CA6/POWER", "ON")
                }
            case "A00A13":
                {
                publish("broker", "cmnd/sonoff-065CA6/POWER", "OFF")
                }
            case "A00A1D":
               {
                publish("broker", "cmnd/sonoff-101AEE/POWER", "ON")
                }
            case "A00A15":
                {
                publish("broker", "cmnd/sonoff-101AEE/POWER", "OFF")
                }                
            case "A00A1E":
               {
                publish("broker", "cmnd/sonoff-5DE6D5/POWER", "ON")
                }
            case "A00A16":
                {
                publish("broker", "cmnd/sonoff-5DE6D5/POWER", "OFF")
                }
        	case "85D5CC":
        	    {
			    Pir433.sendCommand(ON)
			    }                   
            case "150551":
               {
				postUpdate(WohnFluter, ON)
                }
            case "150554":
                {
				postUpdate(WohnFluter, OFF)
                }   
            case "151151":
               {
				postUpdate(SchrankLicht, ON)
                }
            case "151154":
                {
				postUpdate(SchrankLicht, OFF)
                }   
            case "151451":
               {
				postUpdate(SchrankTechni, ON)
                }
            case "151454":
                {
				postUpdate(SchrankTechni, OFF)
                }   
            case "151511":
               {
				postUpdate(Glaskugel, ON)
                }
            case "151514":
                {
				postUpdate(Glaskugel, OFF)
                }   
            case "151541":
               {
				postUpdate(Leselampe, ON)
                }
            case "151544":
                {
				postUpdate(Leselampe, OFF)
                }   
            case "150151":
               {
				postUpdate(SideRechts, ON)
                }
            case "150154":
                {
				postUpdate(SideRechts, OFF)
                }   
            case "151051":
               {
				postUpdate(Salzlampe, ON)
                }
            case "151054":
                {
				postUpdate(Salzlampe, OFF)
                }   
            case "3575C0":
               {
				postUpdate(Schlafzimmer, ON)
                }
            case "35750C":
                {
				postUpdate(Schlafzimmer, OFF)
                }   
    }
 rfData=""
 RfBridge_RfReceive.postUpdate(rfData)
end

mc1271, 2018-07-07:
Dear Jason,
thank you very much, that was exactly what I was looking for. You made my day.
Works like a charm.
All the best 
Mike.

PS.: Grüße aus Hamburg

lionelbarker, 2019-06-11:
> Restarted but nothing has changed...
>
> ```
> 16:27:42 MQT: Subscribe to cmnd/DVES_8F42C0/#
> 16:27:42 MQT: tele/RFB119/INFO1 = {"Module":"Sonoff [...]

Did you ever solve this issue?   I too have a sonoff 433 MHz bridge and all I get is 155F  as the rfcode data

Jason2866, 2019-06-12:
This looks like Intertechno code (false decoded). This happens with orig. EFM firmware.
You have to flash Portisch (Open source alternative) firmware to the EFM chip in the Rf Bridge
How is in wiki described

JulianNaumann, 2020-06-14:
Hi all,
I still have issues learning the RF codes of my remote socket. I have a Sonoff RF Bridge and try to learn the On/Off codes for this socket:
https://www.amazon.de/CSL-Funksteckdosen-Fernbedienung-LEDStatusanzeige-Kindersicherungsschutz/dp/B081K73NY3/
I have the latest Portisch Firmware (0x04) installed on the Bridge, and 8.2.0(tasmota).

Here is the console log:
```
15:56:43 CMD: rfkey2 2
15:56:43 MQT: stat/sonoff/RESULT = {"RfKey2":"Start learning"}
15:56:44 MQT: tele/sonoff/RESULT = {"Time":"2020-06-14T15:56:44","RfRaw":{"Data":"AAA055"}}
15:57:15 MQT: tele/sonoff/RESULT = {"Time":"2020-06-14T15:57:15","RfRaw":{"Data":"AAA255"}}
```

Once I enter rfkey2 2, the bridge beeps, but there is no red LED on, which I thought should be on?
Also there is no "learning failed" message.

I was able to sniff the codes of the remote with the the ```rfraw``` command, but sending the resulting 0xB0 code does not work reliably, only about 10% of the times.

Anybody got any ideas?