arendst/Tasmota #1182: Openhab2 not updating (web or app) when manually toggling sonoff

2e0vet, 2017-11-13:
If I manually toggle the sonoff using the button (or touch pad on sonoff touch), or I I use the "Toggle" button from within the devices web page, then the button status does not update on openhab2.

However if i toggle the switch from withing the openhab2 system, then the the status also toggles between ON/OFF on the devices webpage.

How can I get the ststus of the button to show correctly in openhab2 when I manually toggle the button.

Thanks in advance

davidelang, 2017-11-14:
teach openhab to watch for tele/ and stat/ messages and to not update the status 
when it changes things, only when it gets a message telling openhab that 
something has changed.

something along the lines of:

Switch sw03 "LR Light" { 
mqtt=">[openhab-mqtt:cmnd/sw03/POWER:command:OFF:off],>[openhab-mqtt:cmnd/sw03/POWER:command:ON:on],<[openhab-mqtt:stat/sw03/POWER:state:MAP(binary.map)],<[openhab-mqtt:tele/sw03/POWER:state:MAP(binary.map)]", 
autoupdate="false" }

2e0vet, 2017-11-14:
Thanks for that, unfortunately it didnt work, but that is probably down to me I am very new to Openhab2 and MQTT so finding it a little difficult to debug. I have listed below my items file, 

Switch Magnifier and Switch Garage TV both work OK, in respect of i can turn them on and off.  I copied your "sw3" line and modified it for my MQTT and topic settings.

I'm not sure if this is related but the icon in openhab2 is different see attached screen shot
[image]

Switch Magnifier "Garage Magnifier" <poweroutlet> [ "Switchable" ] {mqtt=">[broker:cmnd/sonoff-7/power:command:OFF:0],>[broker:cmnd/sonoff-7/power:command:ON:1]"}
Switch GarageTV "Garage TV" <poweroutlet> [ "Switchable" ] {mqtt=">[broker:cmnd/sonoff-15/power:command:OFF:0],>[broker:cmnd/sonoff-15/power:command:ON:1]"}
Switch sw03 "Magnifier2" <poweroutlet> [ "Switchable" ] {mqtt=">[broker:cmnd/sonoff-7/POWER:command:OFF:OFF],>[broker:cmnd/sonoff-7/POWER:command:ON:ON],
<[broker:stat/sonoff-7/power:state:MAP(binary.map)],<[broker:tele/sonoff-7/power:state:MAP(binary.map)]",autoupdate="false" }

davidelang, 2017-11-14:
what I posted probably wouldn't work as-is because you probably don't have a map 
file defined exactly the way I do.

go through what I had on the line and look at what each portion is doing. Once 
you understand it, it should be pretty obvious where your config is different.

2e0vet, 2017-11-14:
OK, are you able to point me in the direction of the relevant documentation, as I am finding it quite baffling, that said, I'm pretty chuffed i have managed it this far in under a week

mtijerino78, 2018-01-24:
This is how can resolve the issue, 
in items file this is what I use and works 100 % updating the GUI in my cell phone if I toggle manual in sonoff switch button

Switch Luz_Sonoff_1 "Sonoff1" <light> (PP_Sala, gLuces, Luces, gMyOpenhab) {mqtt=">[localBroker:cmnd/DVES_950A3D/POWER:command:ON:1],>[localBroker:cmnd/DVES_950A3D/POWER:command:OFF:0],<[localBroker:stat/sonoff/POWER:state:default]"}

the part of the line <[localBroker:stat/sonoff/POWER:state:default]

in  /var/log/mosquitto/mosquitto.log use this sudo tail -f mosquitto.log when press the hard button watch the line of command shows in my case mqtt recive this  stat/sonoff/POWER

closed 2018-06-08 by stale[bot]