arendst/Tasmota #1184: v5.9.1c not working with ALEXA, why ?

Olli80, 2017-11-15:
I have tried so many times to connect my sonoff with ALEXA after flashing v5.9.1c ,
but ALEXA didn't find the device !?

Olli80, 2017-11-15:
I have tried WeMo and HUE emulation and ALEXA didn't find anything....

Alveran81, 2017-11-15:
Which Alexa? Echo Plus or Dot?

Olli80, 2017-11-15:
Do I need a Echo or Dot to work with the sonoff ?
At the moment ALEXA is used by FireHD10 Tablet !?

Alveran81, 2017-11-16:
The new Echo with Integrated Hub doesn't work. My old echo dot works.

Olli80, 2017-11-16:
Do I need minimum one Echo at my network,
or did it work with the ALEXA on the FireTablet?

Alveran81, 2017-11-16:
I don't know

AcHoBi, 2017-11-21:
You must have one physical hardware Echo in your local network as a "communication server"!
For sending commands you can use also software Echos like REVERB or the actual Alexa app itself.

Olli80, 2017-11-21:
Thank you, that are good news 👍

closed 2018-05-07 by stale[bot]