arendst/Tasmota #1185: Help to detect failures

smadds, 2017-11-15:
Not a bug, but wondering if there is anything that can be done in the code to flag failures?

Yesterday a DHT22 sensor stopped working on a Wemos based box I was using as a thermostat in a room. Because i did not notice this, the heaters were left on in the room for hours, causing some concern.

I know that within Node red I could do some polling of all the devices, but that seems against the spirit of an event based system like this.

Would it be possible to have a mode that could be enabled that published errors found in Tasmota to MQTT? As far as I can see, if a sensor fails the only outcome is that it stops sending stat updates. It would be good to have a message triggered when a sensor stops working.

davidelang, 2017-11-15:
It would be better to do this in your home automation software, there are lots 
of reasons why you could stop getting data, and a sensor failing in a way that 
Tasmota can detect is a very small fraction of the cases (and it would add a 
fair bit of code that would only be very rarely used, which means that it's hard 
to test as well)

smadds, 2017-11-15:
Not sure I would agree it would be better, David, as it would involve a lot of polling. How about an mqtt publish when a new sensor is found and again when it becomes unavailable? Surely the firmware must be handling this already & it would not take much to push out a publish on change?

CommodoreWhite, 2017-11-17:
Not sure why you'd choose to poll devices to check they're alive. Faced with your concern, I'd kick off a bigtimer every time I received a sensor reading. You get to decide how long you can do without an update. If the timer expires you know there's an issue. You might need a timer for each sensor.
That said, If I had lots of potentially iffy sensors, I'd probably populate an array of due times for each sensor and check it each minute. Hmmm, I think that's called polling. So you were right in the OP.

davidelang, 2017-11-17:
The tasmota is not dynamically discovering sensors, it waits for you to define a 
GPIO as a particular sensor type and then starts reporting what it reads.

There are a lot of sensor failures that show up in different interesting ways

The device is already sending out sensor data every teleperiod, so no extra 
network traffic is needed to watch for it.

closed 2018-05-07 by stale[bot]