arendst/Tasmota #1188: Add pulse count measure for tasmota in some GPIO with some configured divider

roblad, 2017-11-15:
Hi,

Is it possible to add some pulse count measure as an imput GPIO pin for i.e. water or energy pulse measure

i.e.

https://github.com/rogerclarkmelbourne/arduino-esp8266/blob/master/esp8266/cores/esp8266/core_esp8266_wiring_pulse.c

http://www.instructables.com/id/Wearable-heart-beat-sensor-ESP8266Pulse-sensor/

https://learn.openenergymonitor.org/electricity-monitoring/pulse-counting/interrupt-based-pulse-counter

https://community.openenergymonitor.org/t/wemos-d1-mini-esp8266-pulse-counting/4405

davidelang, 2017-11-15:
I believe that something like that was already added.

David Lang

roblad, 2017-11-15:
Hi,

Do U know what is a description in soft for that ?

ionciubotaru, 2017-11-16:
Red about counters https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#sensor

roblad, 2017-11-16:
Hi,

I see

CounterType<x> |   | Show current Counter 1..4 type as pulse Counter or pulse Timer
-- | -- | --
CounterType<x> | 0 / off | Set current Counter 1..4 type as pulse Counter
CounterType<x> | 1 / on | Set current Counter 1..4 type as pulse Timer

but haw to usage, I need to set one GPIO as counter 1...4 and count the value ? I need some divider i.e. for each 50 counts 1 send counted value as 1

davidelang, 2017-11-17:
the sonoff will report the count, you do the math in your home automation 
software

On Thu, 16 Nov 2017, roblad wrote:

> Date: Thu, 16 Nov 2017 10:41:48 +0000 (UTC)
> From: roblad <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f635bf85dd2892ac1982ef45a90b3fb93f021636e92cf0000000116252d6
>     c92a169ce10596c9c@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] Add pulse count measure for tasmota in
>     some GPIO with some configured divider (#1188)
> 
> Hi,
>
>
> I see
>
>
> CounterType<x> |   | Show current Counter 1..4 type as pulse Counter or pulse Timer
> -- | -- | --
> CounterType<x> | 0 / off | Set current Counter 1..4 type as pulse Counter
> CounterType<x> | 1 / on | Set current Counter 1..4 type as pulse Timer
>
>
>
> but haw to usage, I need to set one GPIO as counter 1...4 and count the value ? I need some divider i.e. for each 50 counts 1 send counted value as 1
>
>

myozone, 2017-12-14:
Yes two MQTT outputs would be very useful one for an incremented count and the other time between pulses for a energy or water meter instantaneous watts and Kwh

roblad, 2017-12-14:
see that

https://github.com/stefanbode/Sonoff-Tasmota/issues/15