arendst/Tasmota #1189: Can not flash with Atom

Alveran81, 2017-11-15:
Through the last Update of PlattformIDE in Atom. I can not flash my Sonoffs. I have to Sonoffs and two Laptops. Same error. Can not connect. Before update everything work fine.

stefanbode, 2017-11-16:
If you work on Windows try to remove your .platformio in your home directory. This forces ATOM to rebuild and mostly solves the issue that came with an update. Renaming should also work, if you want to be sure not to detroy something

Alveran81, 2017-11-17:
I removed .platformio directory. And the end i removed all an do a fresh installation. Same Problem.
I also tried a three diferent Operating Systems (Windows 10 x64, Windows 10 x86 and Windows 7 x64)
I dont know whats the Problem.

Joeyhza, 2017-11-17:
I've tried Atom and PlatformIO - great for compiling - had tons of issues flashing.

Now I only use this tool

Flashes first time - every time and a great AIO tool

https://github.com/marcelstoer/nodemcu-pyflasher/releases

Alveran81, 2017-11-18:
Atom is working again on all of my three Computers. It think it was a bug in a Patch

alton76, 2017-11-18:
Give Arduino IDE a see if the issue follows. Most of the time it fix the
issue.

closed 2017-11-18 by Alveran81