arendst/Tasmota #1190: Added support for TYWE3S module as found on SM-PW701E devices

sebixvi, 2017-11-15:
Hello,

I recently bought some WLAN smart sockets on Amazon with type No. SM-PW701E. They use the Tyua TYWE3S module. Datasheet can be found on the net, seems it's an ESP12 device.

Uploading Tasmota went fine, red LED was working right from the start with the Sonoff basic device, but the button was not. So I added the definitions for the GPIO pins as below on the end of the SupportedModules, kNiceList and kModules sections. Now I'm able to use the button as well, the blue LED lights when output is off, when output is switched on, the red LED comes on, the blue one will switch out.

Regards,
Sebastian

`enum SupportedModules {
...,
  SMPW701E,
`
`const uint8_t kNiceList[MAXMODULE] PROGMEM = {
...,
  SMPW701E
};
`
`const mytmplt kModules[MAXMODULE] PROGMEM = {
{...},
{ "SMPW701E",          // SM-PW701E WLAN Socket
     0,
     0,
     0,
     0,
     GPIO_LED1,          //  blue LED
     0,                 // GPIO04 IR or RF receiver (optional)
     0,                 // 
     0,                 //
     0,
     0,
     0,
     0,
     GPIO_REL1,
     GPIO_KEY1,
     0, 0, 0, 0
  }
`

GeoffMaciolek, 2018-01-05:
This same module is used in the SM-SO301 power strip modules, which have 5 relays - 4 for the individual outlets, and one for the group of 4 USB plugs. (I'm not sure if they have any other I/O, power usage, etc). Unfortunately, it looks like flashing these would be a similar level of hassle to the SM-PW701E; there isn't a header of any sort available.

reloxx13, 2018-01-09:
Thx sebi!
Found this board in another cheap smart plug, ill try to flash it.

it was this one: https://www.amazon.de/gp/product/B06XSRTCDF

if it works ill try a pull request.

[image]
[image]
[image]
[image]
[image]
[image]

reloxx13, 2018-01-09:
Would you like to add an image marked the pins to flash?

sebixvi, 2018-01-10:
Am 2018-01-10 00:47, schrieb reloxx13:

> Would you like to add an image marked the pins to flash?

Yes, I can provide an image. Furthermore, I found another chip in
smaller sockets that also supports power metering. I also added that
chip and will post the neccessary configuration. 

As for the power metering: the chip in the other sockets has an unknown
marking (HJL01 if I remember correctly) which I didn´t found. From the
circuit, it seems to be an HLW8012 copy. The resistors for the dividers
are a bit different to the HLW sample circuit (it has two 1.2MOhm
instead of six 470k in the voltage divider), so the readings are way off
with Sonoff default calibration values. Calibration works ok for the
power metering but not for voltage: I calibrated using a 40W light bulb.
When I removed the bulb and plugged in an heater for an aqarium, the
power reading was correct with 75, however, the voltage showed about
400V while the current reading was lower than expected. First I thought
I swapped voltage and current, but when I tried to swap the pin
assignment the readings were totally wrong. 

Maybe there is some other kind of calibration? 

Regards, 

Sebastian

vonBothmer, 2018-01-19:
Wonderful! Just bought one of these devices and it came in the mail today. Sounds like I have a pretty good timing with this..

vonBothmer, 2018-01-20:
Works perfectly but not the easiest to solder..

[image]

[image]

sebixvi, 2018-01-20:
Well, think I'm too slow with my images... ;-) Thx reloxx13 for you picture, the pinout and solder points are like mine.

@vonBothmer: you should consider changeing your wlan password as the original image that was sent with the github mail revealed the config file entry...

vonBothmer, 2018-01-20:
@sebixvi

Shit. The images are mailed also? I edited the post and cropped it out..

ElecTriX666, 2018-01-20:
@sebixvi 

Changed. Please remove reference to my old username.

RrPt, 2018-04-07:
Hello sebixvi,

i hhave an Liminea ZX2820 Switch with power metering. It includes a TYWE3S Module and a HLW8012 power metering IC.  I installed Tasmota and configured Module KMC 70011 since this have same Componets on same Pins.
Switching works well but power meter shows dubios values.
Power ist ok after Calibration, but voltage calibrates with a 25W bulb went to 400V with a 60W bulb. Current ist much too high (ca 10A).
Seems this ist the same issue like u deskribed here on 10.Jan.2018.
Have you solvet this? What was the solution?

Regards Rainer

Kriechi, 2018-04-27:
I have a slightly different socket:
WYSP001B with the TYWE3S module. The PCB is labeled `WIFI_socket_V4_0149` on the back.
I'm using this config now:

```
  { "WYSP001B",
     0, 0, 0, 0,
     0,                // GPIO04 Green LED for WiFi
     GPIO_LED1_INV,        // GPIO05 Red LED
     0, 0, 0, 0, 0, 0,
     GPIO_REL1,        // GPIO12 Relay1
     GPIO_KEY1,        // GPIO13 Button
     0, 0, 0, 0
  }
```

(no voltage or current measurement)

RrPt, 2018-05-06:
Problem was, that on 'ZX2820' sthe signal GPIO_HLW_SEL is inverted compared to  'KMC 70011'.
But this is not configurable.

Opened a new issue:
https://github.com/arendst/Sonoff-Tasmota/issues/2389

Rainer

closed 2018-07-05 by stale[bot]

v01t, 2018-08-12:
Hi @Kriechi,

I have similar WYSP001B with the TYWE3S module on it but with different PCB label: WIFI_switch_0124, I would greatly appreciate if you can help me to find the right pinouts.

[image]

[image]

Kriechi, 2018-08-12:
@v01t I posted my config earlier.
The physical pinout is here: https://github.com/arendst/Sonoff-Tasmota/wiki/TYWE3S and https://docs.tuya.com/en/hardware/WiFi-module/wifi-e3s-module-plug.html

Guzzi-Charlie, 2018-12-04:
Hello Dear all,
this is my first active entry/activity on GitHub after using a lot of tips&tricks during many visits here. I already learned a lot.

I'm using some sonoff and shelly switches within my fhem home automation. All sonoff's I have flashed with tasmota and it's working fine. 

But now I have a (hopefully) small problem.
I also flashed one device (a Luminea SF-500 plug) which uses the TYWE3S modul. I flashed with the newest (6.30) german tasmota version. The flashing procedure was easy and sucessfull as at all others before, but I'm not able to switch the plug. I already tried several settings but without sucess yet.

Is there anyone who has an idea for the correct (tasmota-) settings for this plug?

Guzzi-Charlie, 2018-12-05:
Hello,
in the meanwhile I was able to find some settings to get the plug working.
1. Using the generic module with Button1 on D3 (GPIO00), Relay1 on D5 (GPIO14), LED1 on D7 (GPIO15)
    With this setting I'm able to switch the device via Web-IF and manuall as well, but no power meetering
2. Using the KMC 70011 module
    With this setting switching works fine as well and I receive some power values, but obviously not 
    correct ones.
    [image]

Is there any possibility to calibrate those values? Would be very nice if someone has a solution for this.

RrPt, 2018-12-06:
see my Post from 6. May

Problem was, that on 'ZX2820' sthe signal GPIO_HLW_SEL is inverted compared to 'KMC 70011'.
so current and voltage ist switched, you cant config thit, i have changed in in sourcecode.
But this is not configurable.

Opened a new issue:
#2389

Rainer

Bunny2000, 2019-06-21:
hey guys can the TYWE3S be flashed via rx tx pins like a esp8266 ?

giig982, 2019-06-21:
> hey guys can the TYWE3S be flashed via rx tx pins like a esp8266 ?

yes, they can. You have to find, after flashing, how to map your hw to a known template, anyway.

BreBar07, 2019-07-17:
Hey giig982, seems you might be experienced in reflashing the TYWE3S. I've tried OTA and flashing by serial no luck :(

OTA never detects device.
Serial the flash starts but then fails shortly after.

kozumin, 2019-07-23:
Same for me with another Aliexpress Wifi Dimmer module that comes with TYWE3S.
Trying to put it on boot mode by doing a 3 sec shortcut between GPIO0 and GRND .
It is like, without any shortcut, the led is off. If I shortcut it for 1 sec, the led is lit then goes off. If I shortcut the two pins for 3+ seconds the led goes on and stays on even after removing the shortcut. Which my logic says that is the boot mode but nah, the thing doesn't want to talk to serial when i flash it.
I have 7 of them. I can donate one to someone to hack it and then post the result here. 
[image]
[image]
[image]

clodoaldosouza, 2019-10-21:
In my case, the solution is pullup resistors in RST and another in EN (red) and a pulldown resistor in GPIO15 (black).

[image]

twonky26, 2019-12-21:
The method that worked for me was cutting RX and TX pins on BF7612AM28T chip which is located on the the other side of the board (of Moes DS01C dimmer ). So I guess it will work for any device with TYWE3S which has trouble flashing. 

First, trace RX and TX pins from TYWE3S to BF7612AM28T (or similar) chip.
Second, carefully cut them with sharp blade.
Third, bend them up a little bit, so there is no contact.
After that it should flash just fine. But don't forget to bend them down and resolder after flashing.

[image]
[image]
[image]
[image]
[image]
[image]

mcaden, 2020-01-15:
The solution by @twonky26 did indeed work for me on a MOES DS01C dimmer. Unfortunately when cutting the pins with my exacto knife the pin snapped right at the chip so I'm having a heck of a time trying to put the thing back together if I'll be able to at all. If somebody can find a better way of isolating that chip it'd be much appreciated.

arvidj01, 2020-01-24:
Reading this ... MCU Based Tuya Dimmers and Switches - Flashing - Preparation suggests that holding the reset NRST/RST pin to gnd might allow flashing without having to do any surgery on the MCU pins. However I've not been able to find a BF7612AM28T datasheet to determine which pin is the NRST/RST pin.

If anyone has access to that information would you please share it. I would like to give it a try.

mcaden, 2020-02-04:
I bit the bullet and set up a Pi Zero W to do TuyaConvert.

GijsTimmers, 2020-02-22:
here's how I did it: only connect VCC and GND. that will keep the device powered on. after that you can flash via tuya-convert.