arendst/Tasmota #1195: Rssi 100

flexiti, 2017-11-16:
If anybody have RSSI show at 100 ? after uploading the program.
 I have not found the rule yet, eg. now I have still RSSI 100, upgrade was 2 hours ago

ionciubotaru, 2017-11-17:
RSSI means the wifi signal strength - you get 100 because your sonoff is close to te AP.
Maybe you are looking for uptime.

flexiti, 2017-11-17:
no, this is not that problem, and I know what it is RSSI  ;)

swampdogmash, 2018-04-01:
I have a similar display with some of my units.. one which is next to another which reports only 83 rssi. Not sure what causes it but keen to have a look later this long weekend in the UK. I thought it was a version 5.11 thing but one of these is an S20 I just bought and installed 5.12 on and one is a sonoff basic running 5.11, right next to a sonoff basic running 5.12...

Need to grab some logs and configs and will post back later.

closed 2018-06-20 by stale[bot]