arendst/Tasmota #1199: RF compat with HomeEasy, Chacon, KlikAanKlikUit,...

peturdainn, 2017-11-16:
I'm currently trying to get the Sonoff RF Bridge to work with Home Easy remotes, and of course it doesn't work. So I started investigating.

I already found that somebody decoded the protocol here:
https://blog.domadoo.fr/guides/principe-du-protocole-homeeasy/ 
(warning, in French)

Didn't figure out what chip is used, they are all without markings but it does look like a simple fixed length protocol

I haven't looked in detail, but is there anything indicating this will never work?

mhavo, 2017-12-26:
It should be possible: https://playground.arduino.cc/Code/HomeEasy

But the original Itead firmware does not work neither does the Espurna... So what is so hard with this protocol?

chrisberesford, 2018-01-24:
HomeEasy support would be really useful - these are widely available in Europe under a couple of brands (Nexa being another one).

peturdainn, 2018-05-25:
No you will not, naughty bot.

Bump to keep relevant

ascillato, 2018-05-25:
Sorry, the hardware from Sonoff RF Bridge is not compatible. So Tasmota Firmware can not do anything about that.

peturdainn, 2018-05-25:
Do you have any pointer to doc that shows the difference? All I can see is that home-easy uses the same 3/4 - 1/4 fixed length bit scheme as EV1527 and others. So I'm trying to figure out where exactly the problem lies

Thanks

ascillato, 2018-05-25:
The problem seems to be in the software that is inside the RX chip. It only respond to certain protocols and ignore others.

I don't have now those docs. I need to look for them. I'm writing from what I remember when trying to use others RF 433Mhz controls I have. If the protocol does not match. It is ignored. Sorry. You will need a RX chip for that specific protocol, so that RX chip could send by serial the code received.

Tasmota can not do anything about that. Is Sonoff hardware related.

The Sonoff Bridge chip only allows some protocols.

peturdainn, 2018-05-25:
OK thanks

ascillato, 2018-06-13:
Hi

If your issue was solved, please close it. Thanks :+1:

peturdainn, 2018-06-13:
Well, it is not. But I'm chatting with itead now...

closed 2018-06-13 by peturdainn

norman-knott, 2019-01-26:
Hey here!
Does the bridge not support HomeEasy even with the Portisch firmware?

@peturdainn Any luck with your chat?

Bodge-IT, 2021-10-18:
> Hey here! Does the bridge not support HomeEasy even with the Portisch firmware?
> 
> @peturdainn Any luck with your chat?

I know this is very old but just in case this helps, Brett, one of the founders of the Home Automation Hub had this whole protocol nailed and built into the project.  It was rock solid and working for me until early this year when the Orange hardware failed.
Maybe some of their work is relevant:
http://www.dbzoo.com/livebox/livebox
https://github.com/dbzoo/hah

peturdainn, 2021-10-18:
I followed https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/ and specifically the part about Portisch.
So after flashing Portisch you start the listener: rfraw 177

But first pair the RF device to one button of your RF remote
Then press the button on the remore while Portisch is listening, do that several times. Pick the code that you see multiple times.

Example: in NodeRed the command to switch a light on looks like this:
RfRaw AAB0250308014003B612D4281819081819090818190818190908190818190818190819081909081855; RfRaw 0

This works for my HomeEasy switches