arendst/Tasmota #1200: Wifi Client and AP

Joeyhza, 2017-11-17:
Hi,

Not really an issue at all - but I was wondering :

Because I have a Sonoff SV connected to my main gate motor quite a distance away from my Router which intermittently loses wireless signal, is it all possible to have a nearer Sonoff device function as both a a client and an AP set as a wireless extender/repeater?

Thanks in advance

davidelang, 2017-11-17:
no, the tasmota firmware cannot act as a client and AP at the same time
k

Joeyhza, 2017-11-17:
Yes....okay

I'm quite aware of what the firmware does and does not offer - I was merely wondering about the possibilities 

k

davidelang, 2017-11-17:
well, it's software, so you can do just about anything if you try hard enough 
:-)

but it's unlikely that this will be added to tasmota as we are very tight for 
space. It's better to setup a new AP to cover the area than to try to have the 
sonoff devices relay the messages.

David Lang

  On Fri, 17 Nov 2017, Joeyhza wrote:

> Date: Fri, 17 Nov 2017 19:19:46 +0000 (UTC)
> From: Joeyhza <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24fd156a3ebed3340f0a9825ffe3adb39e30e59eb9292cf000000011626f85
>     292a169ce10613958@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] Wifi Client and AP (#1200)
> 
> Yes....okay
>
> I'm quite aware of what the firmware does and does not offer - I was merely wondering about the possibilities
>
> k
>
>

Joeyhza, 2017-11-17:
Cool - Thanks again David

closed 2018-05-07 by stale[bot]

ionciubotaru, 2020-04-12:
Here is example of wifi repeater.
https://github.com/martin-ger/esp_wifi_repeater
Is it possible to add this extension to Tasmota Lite ?