arendst/Tasmota #1201: Error compiling for board Generic ESP8266 Module.

samiraa78, 2017-11-17:
exit status 1
Error compiling for board Generic ESP8266 Module.

Getting the above error can someone guide me with the solution

samiraa78, 2017-11-17:
Arduino: 1.8.5 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DOUT, 115200, 1M (64K SPIFFS), ck, Disabled, None"

Build options changed, rebuilding all
In file included from C:\Users\samir\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\Robot_Control\src/SquawkSD.h:4:0,

             from C:\Users\samir\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\Robot_Control\src/ArduinoRobot.h:24,

             from C:\Users\samir\Downloads\Sonoff-Tasmota-5.9.1\Sonoff-Tasmota-5.9.1\sonoff\sonoff.ino:1:
C:\Users\samir\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\Robot_Control\src/Fat16.h:27:26: fatal error: avr/pgmspace.h: No such file or directory

#include <avr/pgmspace.h>

                      ^
compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
arduino: 1.8.5 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DOUT, 115200, 1M (64K SPIFFS), ck, Disabled, None"Build options changed, rebuilding allIn file included from C:\Users\samir\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\Robot_Control\src/SquawkSD.h:4:0, from C:\Users\samir\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\Robot_Control\src/ArduinoRobot.h:24, from C:\Users\samir\Downloads\Sonoff-Tasmota-5.9.1\Sonoff-Tasmota-5.9.1\sonoff\sonoff.ino:1:C:\Users\samir\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\Robot_Control\src/Fat16.h:27:26: fatal error: avr/pgmspace.h: No such file or directory #include <avr/pgmspace.h> ^compilation terminated.exit status 1Error compiling for board Generic ESP8266 Module.This report would have more information with"Show verbose output during compilation"option enabled in File -> Preferences.

I managed to get the above error sorted but now i am getting the following error

Snipercaine, 2017-11-17:
I don't know what "Robot_Control" is.  Maybe its something in the user_config.h that you can comment out.

Snipercaine, 2017-11-17:
you could try grabbing an older version of TASMOTA.

arendst, 2017-11-17:
Robot_control shouldn't be there in the first place. It is not used by Tasmota.

Snipercaine, 2017-11-17:
Awesome! Thank you Theo!  

So, I'd say do a fresh install the Arduino IDE in a different folder.  There's probably a more elegant solution to take out the Robot_Control, but I don't know what it is.

samiraa78, 2017-11-17:
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Getting the following error now any idea why

samiraa78, 2017-11-17:
Finally got it working deleted the robot libs and used an old version the last error was bec in a vid i was told you switch tx an cx which was not working clearly

Snipercaine, 2017-11-17:
cool, so you got it working?  Fantastic!

samiraa78, 2017-11-17:
actually when it was connected to the computer it was connecting to wifi now on power supply it does not connect to the wifi network no clue what is the reason any ideas

samiraa78, 2017-11-17:
now i got it to connect to wifi but when i use it automatically turns on when i turn it off from alexa looks like its got its mind of its own

samiraa78, 2017-11-17:
18:36:22 RSLT: stat/sonoff/RESULT = {"POWER":"ON"}
18:36:22 RSLT: stat/sonoff/POWER = ON
18:36:23 MQTT: Attempting connection...
18:36:24 mDNS: Query done with 0 mqtt services found
18:36:30 MQTT: Connect FAILED to domus1:1883, rc -2. Retry in 10 seconds
18:36:31 UPnP: Multicast (re)joined
18:36:41 MQTT: Attempting connection...
18:36:42 mDNS: Query done with 0 mqtt services found
18:36:48 MQTT: Connect FAILED to domus1:1883, rc -2. Retry in 10 seconds
18:36:49 UPnP: Multicast (re)joined
18:36:59 MQTT: Attempting connection...
18:37:00 mDNS: Query done with 0 mqtt services found
18:37:07 MQTT: Connect FAILED to domus1:1883, rc -2. Retry in 10 seconds
18:37:07 UPnP: Multicast (re)joined
18:37:17 MQTT: Attempting connection...
18:37:18 mDNS: Query done with 0 mqtt services found
18:37:25 MQTT: Connect FAILED to domus1:1883, rc -2. Retry in 10 seconds
18:37:26 UPnP: Multicast (re)joined
18:37:27 RSLT: stat/sonoff/RESULT = {"POWER":"OFF"}
18:37:27 RSLT: stat/sonoff/POWER = OFF
18:37:36 MQTT: Attempting connection...
18:37:37 mDNS: Query done with 0 mqtt services found
18:37:44 MQTT: Connect FAILED to domus1:1883, rc -2. Retry in 10 seconds
18:37:45 UPnP: Multicast (re)joined

this is the log it wont connect to MQTT as i do not have a MQTT server setup 
But can anyone guide me as why i am getting issues with Alex it always gives me a error device unresponsive and whenever i turn it off it automatically turns the device on after few seconds

cclckc, 2018-01-25:
In file included from R:\19 sonoff\Sonoff-Tasmota-5.11.1\Sonoff-Tasmota-5.11.1\sonoff\sonoff.ino:37:0:

sketch\sonoff_post.h:106:143: fatal error: core_version.h: No such file or directory

 #include <core_version.h>                   // Arduino_Esp8266 version information (ARDUINO_ESP8266_RELEASE and ARDUINO_ESP8266_RELEASE_2_3_0)

                                                                                                                                               ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

Snipercaine, 2018-01-25:
Sorry man.  I don't have a solution for that.  Maybe someone smarter than me can weigh in.

per1234, 2018-02-08:
That line was commented out in https://github.com/arendst/Sonoff-Tasmota/commit/d66ef2347e7fb9f597fa4834dbca89c4de21cc79 so the issue should be resolved.