arendst/Tasmota #1206: error during compiling project

ericsboro, 2017-11-17:
Hi Theo,

I've done everything according to your instructions but this time compilation failing.
Compilation of v.5.8.0q completes with no errors.

Kindly advise what could be the problem? 

xArduino: 1.8.5 (Mac OS X), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DOUT, 115200, 1M (no SPIFFS), ck, Disabled, None"

Archiving built core (caching) in: /var/folders/9l/1nkny48n3n552fwmfqkz693c0000gn/T/arduino_cache_540007/core/core_esp8266_esp8266_generic_UploadTool_esptool,CpuFrequency_80,FlashFreq_40,FlashMode_dout,UploadSpeed_115200,FlashSize_1M0,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____abbd1029fed3b39a8955cea9747ed5b5.a
/Users/rozum/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sonoff.ino.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

arendst, 2017-11-17:
Disable ws2812 feature in user_config.h and try again. I noticed the use of iram1_0_seg is close to the max and some features go over it.

Using VisualStudioCode for some reason allows just more of that tiny amount of iram use.

ericsboro, 2017-11-17:
Thanks a ton!!! it's working now!

closed 2017-11-17 by ericsboro

EDsteve, 2018-03-09:
Hi,
I have the same error message but where can i disable ws2812 in user_config.h?
I can't find this file anywhere.