arendst/Tasmota #1208: SDA / SCL pins for I2C communication

yaydogan, 2017-11-17:
Hi,

I have a Sonoff and looking at the ESP8266 pinout  diagram, I see GPIO4 is SDA and GPIO5 is SCL. I want to use these I/O pins for an I2C OLED and soldered wires on the chips. However these bits are defined as:
I2C_SCL - 5
I2C_SDA - 6 in Tasmota software. Has anyone used I2C interface with an OLED and did it work? Or do I need to change the values of SCL, SDA pins to use with OLED display?

Thanks.

arendst, 2017-11-17:
Tasmota does as yet not support OLED or any display.

webtop, 2018-01-19:
I'm going to be working on some I2C code this weekend. If I come across anything that I think will help you, I'll let you know, but no guarantees,

closed 2018-06-02 by stale[bot]