arendst/Tasmota #1209: Sonor 5V USb Module pulse loop

freaktech, 2017-11-17:
hi i git some 5 Volt USB sonoffs

flashed 1  100%
get access to the web interface
but after the first switch it starts to switch on of in a loop

i set the module  to Sonoff SV in the web interface

any ideas?

thanks
tom

ionciubotaru, 2017-11-17:
you selected a wrong module - the relay is on the module's led pin, that's why it switch on and off

freaktech, 2017-11-17:
?!? and what must i select for this module?

https://www.ebay.com/i/282733639698?rt=nc

freaktech, 2017-11-18:
ok i got it

i must set to basic, but the outside switch must be pressed once LED on
settings will be stored of this button after power loss

so looks like its working now 
also when i unplug it

thx

closed 2018-05-07 by stale[bot]