arendst/Tasmota #1211: What unit is Uptime?

TommySharpNZ, 2017-11-17:
My POW reports it's uptime as "163"..... I can't seem to determine what this is without rebooting it to test. I'm guessing hours?

TommySharpNZ, 2017-11-17:
Should have checked the web interface..... Looks like it is hours....
Is it possible to get minutes? I'd like to troubleshoot my 4CH rebooting by monitoring how long it is "up" for between reboots.

zerog2k, 2017-11-20:
It's in hours, but something which might be useful is capturing the startup timestamp, and emitting that in the rtc status message ("Status 7")

JoergZ2, 2017-12-23:
Is there no option to get the exact uptime as a timestamp or as running time with hh:mm:ss?

davidelang, 2017-12-23:
there is not, it's only in hours

kiwijunglist, 2018-02-26:
My uptime is:
1T18:41:49

is that 1 day 18hr 41min 49s ??

If it is, then it's surprising because I installed the sonoff maybe 1 week ago, so it must be having rebooting problems.  Does uptime reset if I take my MQTT server offline?  Does it reset if I take my WiFi offline?

zerog2k, 2018-02-26:
Yes, the format was recently updated.

> 1T18:41:49
> is that 1 day 18hr 41min 49s ??
I believe so.