arendst/Tasmota #1216: Switch<x>-Relay<x> assignment not correct

boaschti, 2017-11-18:
I tried few configs and not all works correct.
{noformat}
GPIO2   LED1i
GPIO12  Relay1
GPIO16   Switch1

The Relay1 has the toggled state after Restart.
see below with other extenal Parts.

GPIO2   LED1i
GPIO12  Relay1
GPIO13   Switch1

OK.

GPIO2   LED1i
GPIO12  Relay2
GPIO13  Relay1
GPIO14   LED2i

Led is on if Relay1 OR Relay2 is on. Last Relay State is present after restart.

GPIO2   LED1i
GPIO12  Relay2
GPIO13  Relay1

Led is on if Relay1 OR Relay2 is on. Last Relay State is present after restart.

GPIO2   LED1i
GPIO12  Relay2
GPIO13  Relay1
GPIO16  Button1

Sonoff is resets its config.

GPIO2   LED1i
GPIO5   Switch2
GPIO12  Relay1
GPIO13  Relay2
GPIO16  Switch1

SwitchMode1 = 1

Led is on if Relay1 OR Relay2 is on. Everything works.

GPIO2   LED1i
GPIO5   Switch2
GPIO12  Relay1
GPIO16  Switch1

SwitchMode1 = 1

Relay1 and Led1 is on if Switch1 OR Sw is on.
{noformat}

davidelang, 2017-11-18:
see poweronstate

also see the wiki page on poweronstate and retain, you may have some retained 
messages messing you up.

turn on 'more debugging' in the logs and see what is happening at boot.

boaschti, 2017-11-18:
Hi, 
{"PowerOnState":3}
{"PowerRetain":"OFF"}
I have no retained messages.

I check this again with more logging.

boaschti, 2017-11-26:
I don't tetsed it yet but do you have a idea what happens in the rest of configs?

boaschti, 2017-11-27:
I tried "more debug" for web browser. There are no messages shown at the time where switch is pressed/toggled. There are no messages from mqtt.
I updated the upper text. So why switch1 and switch2 is linked if only one relay is configured? I understand that switch1 is linked to relay1 and led1, switch2 is linked to relay2 and led2. Is that right?
Do i always need a relay for a configured switch?

closed 2018-05-07 by stale[bot]