arendst/Tasmota #1217: compile error: bits/c++config.h

danialFACT, 2017-11-19:
hello, trying to compile and upload and i'm getting this error:

Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DOUT, 115200, 1M (64K SPIFFS), ck, Disabled, None"

In file included from C:\Users\D\Desktop\arduino-1.8.5\portable\sketchbook\libraries\PubSubClient\src/PubSubClient.h:77:0,

                 from C:\Users\D\Desktop\arduino-1.8.5\portable\sketchbook\sonoff\sonoff.ino:39:

c:\users\D\Desktop\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional:48:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>

                            ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

niklaskemi, 2017-11-21:
I have the same problem. How to solve this?

arendst, 2017-11-21:
Cannot reproduce. Make sure to follow the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

danialFACT, 2017-11-21:
Hello Theo, I will close the issue as, (as usual,) it was a user error - I had only installed the ArduinoJSON and ESP8266 libraries (as was described in a number of youtube tutorials...) and not ALL the other dependencies you include in the folder in your download.

closed 2017-11-21 by danialFACT

insanity67, 2018-01-16:
Had this when my path to the arduino portable was too long. Shortened the path names, and everything compiled as should.

danialFACT, 2018-04-12:
@NicoSoftware - there are a number of folders included in the zip download of TASMOTA - move them to the relevant places in your Arduino folder and it should sort you out.

amitubale, 2018-05-10:
i deleted all the old ardunio and tasmota release code and followed the instruction in Wiki..download and upziped from scratch, moved all the filed needed to corresponding folders......getting the same c++ fatal error. Not sure what to do..please help.

amitubale, 2018-05-10:
here are the error details
arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:60:0,
rduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:60:0,
arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\utility:68:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>

                            ^
compilation terminated.

amitubale, 2018-05-11:
ok resolved the issue after reverting to older version on esp8266 board manager package 2.3.0.
Complied without any errors. teh Tasmota Wiki indicated for 2.3.0 to set Flash Size 1M (no SPIFFS) -- that option was not available so used 1M( 64,...and it seems to be working ok.Tested a couple sonoffs and its functioing. I hope some of the admins can update the Wiki for the newer version of packages. thanks