arendst/Tasmota #1220: Add plain esp-01 and esp-12x to sonoff_template

MatzeJoerling, 2017-11-19:
Hi,

Feature Request:
im setting up a bunch of esp-01 and esp-12 modules and it would be great to have them as module type in  sonoff_template .
Is that possible?

regards Matze

arendst, 2017-11-19:
esp-01 with 0.5 M flash won't work.
esp-12 is wemos and is available out of the box.

MatzeJoerling, 2017-11-19:
esp-01 has 1M flash, have it running (i also soldered a 4M on esp-01 for testing :-) )
wemos, ok , but would be nice to have esp-12 shown... 
tnx for answer

KPK75, 2017-11-30:
Just modify sonoff_template.h
Some ESP-01 ( e.g. with black PCB from Robotdyn ) have 1M flash. Also "ESP-01S" have 1M.

benzino77, 2017-12-22:
The black modules has 1M flash (AFAIK). OTA works fine. 
@arendst maybe you can add native support for this module?
```
...
// Supported hardware modules
enum SupportedModules {
  SONOFF_BASIC,
  SONOFF_RF,
  SONOFF_SV,
  SONOFF_TH,
  SONOFF_DUAL,
...
  SONOFF_DUAL_R2,
  ESP_01,
  MAXMODULE };
```
then

```
const uint8_t kNiceList[MAXMODULE] PROGMEM = {
  SONOFF_BASIC,
  SONOFF_RF,
  SONOFF_TH,
  SONOFF_DUAL,
...
  WEMOS,
  WITTY,
  ESP_01
};
```
and
```
const mytmplt kModules[MAXMODULE] PROGMEM = {
  { "Sonoff Basic",    // Sonoff Basic (ESP8266)
     GPIO_KEY1,        // GPIO00 Button
...
  { "ESP 01",  // Sonoff Dual R2 (ESP8285)
     GPIO_USER,        // GPIO00 Button 0 on header (0 = On, 1 = Off)
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_USER,
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     0,
     0,        // GPIO05 Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     0,        // GPIO09 Button 1 on header (0 = On, 1 = Off)
     0,        // GPIO10 Button on casing
     0,                // Flash connection
     0,        // GPIO12 Relay 1 (0 = Off, 1 = On)
     0,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  }
};
```
Should do the job I think.

stoehrmark, 2018-01-11:
DIY SONOFF using 
ESP-01 (black 1M), Sonoff-Tasmota 5.10.0e and above modifications (@benzino77 Thank You) 
All works perfectly including OTA

Memory Info after uploading. From "Information" Page
Flash Size 1024kB
Program Flash Size	1024kB
Program Size	445kB
Free Program Space 556kB
Free Memory	27kB

@arendst this would make a good addition for DIY or more advanced builds.

https://photos.app.goo.gl/mV66qkgRs6KsL4RE3

Thank you for the great firmware!!!!

davidelang, 2018-01-11:
the wemos module type already lets you define every GPIO, so there's no need to 
add many other configurable types.

FanLin-ffms, 2018-01-14:
The WeMos is a nice module but it is a fat bumblebee comparing the ESP-01.

davidelang, 2018-01-14:
On Sun, 14 Jan 2018, Arnim Faryn wrote:

> The WeMos is a nice module but it is a fat bumblebee comparing the ESP-01.

So?

The only thing that matters as far as the firmware is concerned is that all 
GPIOs are user defined. This makes it a generic setting that works for any 
ESP8266 based device.

stoehrmark, 2018-01-14:
I agree that the WeMos setting will work perfectly for the ESP-01. But for a new person, this may be confusing. There are a lot of people getting into Home Automation that do not have a full understanding of what they are doing. Adding the ESP-01 to the list will just improve the ease of use of the firmware. The easier the better.

Once again Thank You for all your hard work on this project.

davidelang, 2018-01-14:
e can't add every possible device to the list due to space being tight, but I 
agree that renaming the wemos setting to generic (or something like that) would 
be a good idea

closed 2018-05-28 by stale[bot]

alexandre-shima, 2019-01-20:
@stoehrmark I'm having issues apparently with the same device, an ESP-01 with 1 MB (black board).
Are you using exactly the same text for the changes on the sonoff_template.h file as shown by @benzino77?
I'm getting this error while compiling:
sonoff_template.h:1795:1: error: too many initializers for 'const mytmplt {aka const MYTMPLT}'
If I use the "Sonoff Basic" as a template, changing the first 4 initializers to GPIO_USER and all the rest to 0, the compilation works fine, but the device doesn't work after uploading the code.
Any thoughts?