arendst/Tasmota #1221: SwitchMode question for toggling wall switches

danielsoar, 2017-11-19:
Hi Guys,  Love Tasmota but have an issue.

I have played with switchmode and currently have it set to:

#define SWITCH_MODE            FOLLOW    

the doco in [switchmode doco](https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic) has not been updated to include pushbuttonhold/_INV but from reading in here I think that just includes a long press for second button input back to MQTT to perform a second action.

My issue is that I have physical toggle switches around the house connected to the sonoff boxes and the mqtt can toggle a state change on the relay from any state the physical switch was in, but to get the lights to turn on from the switch again once its changed state from software if the software does not match the switch state anymore I need to flick the switch twice to get it to 'catch up' again.  Bad terminology I know but hopefully you get my point.

This is not helping the WAF (Wife acceptance factor) of my home automation system.  Since the home automation system knows which state it is in, is it possible to turn a toggle switch into a pushbutton by monitoring the edge change high to low OR low to high and using that as a PB? 

I really didn't want to change all the switches in the house to bell press if possible as I have custom slim wall plates that fit on architraves so I wanted to leave the switches as they were if possible for aesthetic reasons.  

I was hoping that either this is already implemented and I have just missed how to achieve it?

If not implemented yet, perhaps it is not too complex to implement and it would help alot of people who want to use this for lighting allowing them to use their current hardware switches assigned to a GPIO as a pushbutton that does edge detection.

Really appreciate the help and sorry if this is a double up question.

kenmcc, 2017-11-20:
I think you and I are trying to do something very similar:

https://github.com/arendst/Sonoff-Tasmota/issues/1194

arendst, 2017-11-21:
How about ``switchmode 0`` (Toggle).

kenmcc, 2017-11-21:
i think the issue with toggle is that his switches are non retractive, so you flip it up to turn on, flip it down to turn off, like a normal light switch. The buttons on the unit though are retractive, and they expect you to close the contact to turn on the relay, and again close the contact to turn off the relay.

So to turn off the relay, he has to firstly flip (back to off i guess) the switch to break the circuit, then flip it back on again at which point the contact is closed and the relay toggles again.

What he's looking for i think is to detect a state change of the switch, either from open to closed, or from closed to open

arendst, 2017-11-21:
How about ``switchmode 3`` (Pushbutton)

danielsoar, 2017-11-22:
Yeah, pushbutton is looking for a momentary low to high signal to change the state of the relay, if i used that i would have to flick my switch twice to get a change. I am looking for a solution to flick the switch once for a state change (monitors and uses both low to high AND high to low as a pushbutton input)

arendst, 2017-11-22:
Try the latest version 5.9.1h which provides ``switchmode 7``. I think this is what you were looking for.

danielsoar, 2017-11-26:
So the concept is right but its still buggy.

* Add option PUSHBUTTON_TOGGLE (SwitchMode 7) to allow toggling on any switch change (#1221)

The switch works and the software works but they dont work together.

If I use switchmode 7 
switch off = relay on
switch on = relay off
software toggle on with switch on and the relay goes on.
BUT if i now turn the switch off nothing happens, I then need to toggle the switch again (turn it on) to get the relay to change state.

So the software works regardless of the switch state but the switch only works if the software left it in the correct follow state.  I really just want the switch rising or falling edge to toggle the relay state.

danielsoar, 2017-11-27:
having slept on it overnight I am convinced that the intention of switch-mode 7 is described correctly in the change notes.  

So either I don't have it turned on, or there is a small bug in the edge detection logic using that as a toggle.

Would it be helpful for anyone else to **feature request that we could select the switch-mode from the web interface?**

I have reflashed my sonoffs (basic, 4chan-pro, pow) with 5.9.1i and they kept their names and settings, although they show the updated firmware version in the information tab.  

Is it possible that even though I selected mode 7 in the config, when I re-flashed they retained their old switchmode?  I ask because they kept some of their other settings after the update (friendly name etc)

How is the best way to check what switch-mode they are currently in - and yes, can we (pretty please) get it to be changeable via the web interface perhaps under where we allocate the GPIO to a particular switch.

so ie: GPIO4 -select- Pushbutton 1
switchmode -select  7 PUSHBUTTON_TOGGLE

Maybe even for completism switch topic could be selectable after this? (or put it on the MQTT page)

At the very least, if there is some technical reason why selecting the mode on the fly is hard, could we have the currently selected switchmode (and topic) displayed on the information page of the web interface to make it easy to check/verify?

Really appreciate your time looking into this @arendst your firmware for these cheap switches is awesome!

-Dan

davidelang, 2017-11-27:
see the troubleshooting page about why settings that are compiled in aren't 
changed after you flash the device with a new image

danielsoar, 2017-11-27:
That is helpful thankyou I will try holding the button down or sending a mqtt reset value tonight after work - do you know if there is any way to check what the current setting is running on the unit (for switchmode)
It was more a suspicion than a statement, but I am unsure how to check.

arendst, 2017-11-27:
Read the wiki and learn about commands used to configure your system without reflashinf or changing almost nothing in user_config.h

Use the web clnsole to enter commands.

Welcome in the 2017s...

puppetmaster886, 2018-01-27:
I there any difference between switchmode 7 and switchmode 0?
Even the code looks the same to me...

      if (button != lastwallswitch[i]) {
.....
        switch (Settings.switchmode[i]) {
        case TOGGLE:
          switchflag = 2;                // Toggle
          break;
......
        case PUSHBUTTON_TOGGLE:
          if (button != lastwallswitch[i]) {
            switchflag = 2;              // Toggle with any pushbutton change
          }
          break;

davidelang, 2018-01-28:
switchmode 0 is for a momentary switch, press it and it toggles, nothing happens 
when you release it

switchmode 7 is an on off switch, switch it on and it toggles, switch it off and 
it toggles again

majherek, 2018-02-16:
Hi,

IMHO PUSHBUTTON_TOGGLE doesn't work on my sonoff basic. A make modification like this: http://support.iteadstudio.com/support/discussions/topics/11000003087/page/last. And connect wall switch to additional connector (blue one). 

When I set PUSHBUTTON_TOGGLE it keeps like SwitchMode = 0 -> it change state only whe the blacku push button in pushed. When I release it, the state dosn't change.

DNA99, 2018-02-25:
Is it possible to have the sonoff power an LED indicator light (when the lights are switched off) so that it will indicate the lights are switched off and also act as a night light to show where the switches are?