arendst/Tasmota #1222: Multiple DS18B20

Amadeus71rus, 2017-11-20:
Hello. How i can set resolution ds18b20 to 11bit? TempRes not working.

ionciubotaru, 2017-11-21:
you can't, it's a digital sensor, not an analog one.

Amadeus71rus, 2017-11-21:
This is not true. The DS18B20 has a resolution of 9.10 and 11 and 12 bit. In normal arduino set by the command setResolution.

arendst, 2017-11-21:
According to the datasheet:

> The resolution of the temperature sensor is user-configurable to 9, 10, 11, or 12 bits, corresponding to increments of 0.5°C, 0.25°C, 0.125°C, and 0.0625°C, respectively. The default resolution at power-up is 12-bit.

So what more do you want?

Amadeus71rus, 2017-11-21:
When using TempRes 3, I only add zeros. The temperature continues to be measured to within 0.5 degrees. As if the accuracy of 9 bits is set.

arendst, 2017-11-21:
It also depends on your DS18x20 sensor. The S version does not have that many resolution.

Amadeus71rus, 2017-11-21:
The sensors are defined as DS18B20. On firmware EspEasy all works correctly.

arendst, 2017-11-21:
Then I suggest you set the DS18B20 resolution in ESPEasy to 12 bits and then move it over to Tasmota. Here is my result:
[image]

Amadeus71rus, 2017-11-21:
[image]

Amadeus71rus, 2017-11-21:
I'll try it.

arendst, 2017-11-24:
Next (pre)release will set resolution to 12-bit if not yet set.

mkh595, 2017-11-25:
_you can't, it's a digital sensor, not an analog one._
@ionciubotaru : You very often write nonsense

ionciubotaru, 2017-11-26:
@mkh595 - thank you for sharing your opinion. I try to answer as many issues as I can, of course sometimes I make mistakes. Please try to answer as many issues as you can - this is a better way to help the community.

closed 2018-05-07 by stale[bot]