arendst/Tasmota #1223: Sonoff Basic, v5.9.1 when toggled off turns back on

scottfuhrman, 2017-11-20:
Have a recently flashed Sonoff Basic. When I toggle the relay off either via the web UI or Alexa, the relay shuts off then turns back on a moment later.  This occurs with MQTT both enabled and disabled.

When Mqtt is enabled:

18:53:59 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
18:53:59 MQT: stat/sonoff/POWER = ON
18:54:00 MQT: stat/sonoff/RESULT = {"POWER":"OFF"} 
18:54:00 MQT: stat/sonoff/POWER = OFF
18:54:02 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
18:54:02 MQT: stat/sonoff/POWER = ON
18:54:05 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
18:54:05 MQT: stat/sonoff/POWER = ON
18:54:09 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
18:54:09 MQT: stat/sonoff/POWER = ON
18:54:12 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
18:54:12 MQT: stat/sonoff/POWER = ON

When Mqtt is disabled:

18:55:59 RSL: RESULT = {"POWER":"ON"}
18:55:59 RSL: POWER = ON
18:56:10 RSL: RESULT = {"POWER":"OFF"}
18:56:10 RSL: POWER = OFF
18:56:14 RSL: RESULT = {"POWER":"ON"}
18:56:14 RSL: POWER = ON
18:56:16 RSL: RESULT = {"POWER":"OFF"}
18:56:16 RSL: POWER = OFF
18:56:17 RSL: RESULT = {"POWER":"ON"}
18:56:17 RSL: POWER = ON

Why is the relay being commanded to turn back on after I turn it off?

scottfuhrman, 2017-11-20:
Seems when I disable WeMo emulation the relay toggles properly.

scottfuhrman, 2017-11-21:
Definitely has something to do with WeMo emulation enabled. When I disable WeMo emulation and enable Hue emulation it works properly.

When WeMo emulation was enabled and I turned the device off with the Echo or Alexa ios App, the device would turn off (and back on) and the Alexa app would say "device not responding".

sbryk, 2017-11-29:
I can confirm this behavoir

erik-wendt, 2017-12-13:
same here! But Hue emulation works as expected

reloxx13, 2017-12-15:
Heya, since my pull request did not get merged till now, u can get my WeMo fix from my fork: https://github.com/reloxx13/Sonoff-Tasmota/tree/feature/Fix_WeMo_Emulation

Alexa i requesting the status of the modul, which turned the device back on.