arendst/Tasmota #1224: h801 and mqtt commands

martinlindjarv, 2017-11-20:
Hello!

I have a H801 wifi controller with Sonoff-Tasmota and i have some issues or problems with commands...

If PWM is used all seems to be working.
switches/cmnd/h801/pwm1 1
switches/stat/h801/RESULT {"PWM":{"PWM1":1, "PWM2":0, "PWM3":0}}

But if i try to use fade or color or almost anything else it says command unknown:
switches/cmnd/h801/color (null)
switches/stat/h801/RESULT {"Command":"Unknown"}
switches/cmnd/h801/fade (null)
switches/stat/h801/RESULT {"Command":"Unknown"}

Version of firmware:
Program Version | 5.9.1
Build Date & Time | 2017-11-07T16:23:04
Core/SDK Version | 2_3_0/1.5.3(aec24ac9)

And as far as i know module type is correct:
Module type (Sonoff Basic)
But from dropdown -> 20 H801

Am i doing something wrong or should i test older firmware?

P.S Excellent work with Tasmota firmware!

Martin

arendst, 2017-11-21:
Use ``setoption15 1`` to select color/dimmer instead of direct pwm control.

martinlindjarv, 2017-11-21:
Excellent, got it working!

Thank you!

closed 2017-11-21 by martinlindjarv