arendst/Tasmota #1225: Help to turnoff status LED on Sonoff Touch

Ftosam, 2017-11-21:
I may need to preface this by explaining I have absolutely zero programming knowledge (although am quite tech savvy).

I bought a Sonoff touch for our bedroom but the status LED is on all night. Drives my missus crazy! (and in turn drives me crazy). 
Is there a way to have the wifi status LED only light up when the relay is on? Or if easier, just disable it all together? 

Any help would be awesome, and I apologise if I don't understand alot of your explanations.

gajotnt, 2017-11-21:
Hello there, im not at home so i cannot try this and give you accurate information but go to the device console page and try these commands:

LedState |   | Show current led state as 0 to 7
LedState | 0 / off | Disable use of LED as much as possible
LedState | 1 / on | Show power state on led (inverted for Sonoff Touch)
LedState | 2 | Show MQTT subscriptions as a led blink
LedState | 3 | Show power state and MQTT subscriptions as a led blink
LedState | 4 | Show MQTT publications as a led blink
LedState | 5 | Show power state and MQTT publications as a led blink
LedState | 6 | Show all MQTT messages as a led blink
LedState | 7 | Show power state and MQTT messages as a led blink

Also to save some power use this:
https://github.com/arendst/Sonoff-Tasmota/wiki/Energy-Saving

gajotnt, 2017-11-21:
And maybe these commands will help also:

LedPower |   | Show current led power state as On or Off
LedPower | 0 / off | Turn led AND LedState Off
LedPower | 1 / on | Turn led On AND LedState Off

closed 2018-05-07 by stale[bot]

reboucasaraujo, 2024-03-29:
Not a "nice fix" but that's what I did after waiting years and sonoff never releasing a firmware to let us enable/disable status LED: just open the front cover, take a very small and pointy tool (it can be a precision screwdriver or thin metal tweezers) and literally just break each one of the three switch status LEDs. They're very fragile actually, and you don't even have to remove the white stickers; you just fit the screwdriver through a gap between the sticker and the board and "poke" the tiny LED. I did this to 4 Sonoff TXT0US (3 switches), and they are working perfectly. I didn't even have to uninstall them from the wall to do it; you can do it with them in place.

[image]