arendst/Tasmota #1226: Clock Improvements

AcHoBi, 2017-11-21:
[image]

I have put a Neopixel clock on the backside of an existing 60cm diameter clock. There are 110 pixels.
1. Because my pixels count is not a factor of 60 the default calculation causes an override of seconds and minutes hand. That means that the seconds hands moves over the 12 o’clock position to the 1 position and suddenly jumps back to the 12 position. I have fixed this by a more accurate calculation:

Changes in xdrv_ws2812.ino V5.9.1f

`void Ws2812Clock()
{
  strip->ClearTo(0); // Reset strip
  // int clksize = 600 / (int)Settings.light_pixels;
  // Ws2812UpdateHand((RtcTime.second * 10) / clksize, WS_SECOND);
  // Ws2812UpdateHand((RtcTime.minute * 10) / clksize, WS_MINUTE);
  // Ws2812UpdateHand(((RtcTime.hour % 12) * (50 / clksize)) + ((RtcTime.minute * 10) / (12 * clksize)), WS_HOUR);
  int clksize = 60000 / (int)Settings.light_pixels;
  Ws2812UpdateHand((RtcTime.second * 1000) / clksize, WS_SECOND);
  Ws2812UpdateHand((RtcTime.minute * 1000) / clksize, WS_MINUTE);
  Ws2812UpdateHand(((RtcTime.hour % 12) * (5000 / clksize)) + ((RtcTime.minute * 1000) / (12 * clksize)), WS_HOUR);

  Ws2812StripShow();
}`

2. Width range of hands is from 1 – 30
For dimming the outer hand pixels a shiftleft is used. 
First Problem was exceeding the uint8 range by using more than 15 ??? pixels for the width. This causes some white bright pixels at several regions of the clock. 
Second Problem was the strong dimming so that only a maximum of 5-7 pixels are visible depending to the overall dimming. So I disabled the dimming for correct 

Changes in xdrv_ws2812.ino V5.9.1f

`void Ws2812UpdatePixelColor(int position, struct WsColor hand_color, uint8_t hand)
{
#if (USE_WS2812_CTYPE > 1)
  RgbwColor color;
#else
  RgbColor color;
#endif

  uint16_t mod_position = mod(position, (int)Settings.light_pixels);

  color = strip->GetPixelColor(mod_position);
  float dimmer = 100 / (float)Settings.light_dimmer;
  //  uint8_t offset = 1 << hand;
  uint8_t offset = 1
  color.R = cmin(color.R + ((hand_color.red / dimmer) / offset), 255);
  color.G = cmin(color.G + ((hand_color.green / dimmer) / offset), 255);
  color.B = cmin(color.B + ((hand_color.blue / dimmer) / offset), 255);
  strip->SetPixelColor(mod_position, color);
}`

Sorry, I wanted to format the code differences a little bit better, but is is my first try here.
Would be great if these changes could be overtaken to the official version ;-)

arendst, 2017-11-21:
Thnx. Will look over it and probably integrate in next (pre) release.

jhlweb, 2017-11-22:
@AcHoBi good improvement.  makes it much better with a 144LED count to. thx.

closed 2017-11-23 by arendst