arendst/Tasmota #1228: Sonoff Touch Alternatives?

gajotnt, 2017-11-21:
Hello there, sorry this isnt directly related to Tasmota, but maybe some of you know something, I live in Europe, and ITEAD has only released the sonoff T1 touchs for UK sockets. I sent them an email asking if there was any date, and they replied:

> Thanks for your attention and we will realease T1 EU 1-2gang at Q1-Q2 2018.
> No 3 gangs at present.

Soo...does anyone know any good alternatives that are programable with at least 1mb flash (For use with Tasmota fw), dont want/need all that RF mumbojumbo :P

Thanks

kevinshane, 2017-11-22:
I am also interested in getting a sonoff like device with good quality, 4 gang version please

gajotnt, 2017-11-22:
I dont think its possible to fit 4 gang (4 relays) in such a small space :/

kevinshane, 2017-11-23:
@gajotnt hey man, I do have a 4 gang sonoff like device, but the quality is .... **really bad** (messy pcb, getting too hot when running a few days... ), so finally I give up ..

ageorgios, 2017-11-24:
ordered this:
https://www.aliexpress.com/item/EWelink-2-Gang-1-way-Wireless-Wifi-Control-Light-Switches-Wall-Touch-Switch-App-Phone-Remote/32830632218.html

not delivered yet.

zarss, 2017-11-25:
@gajotnt Hammer and chisel are your friends. All it is, square against round.

gajotnt, 2017-11-25:
LoL true, could take that route, just sick of concrete dust in my house, if the one I ordered and the other ppl ordered aren't good...then chisel will have to do

kevinshane, 2017-11-27:
@ageorgios I have the exact same type of device, 4gang version though, however the pcb quality is really really bad, but please test it with a long time(at least running 1month, check the heat issues for the sake of safety), maybe I just got a very bad one, the one I ordered is getting very hot after running few weeks(feels like it's ready to fire....) so.. mine is eating dirt currently T_T

gajotnt, 2017-11-28:
http://sonoff.itead.cc/en/products/residential/sonoff-t1-eu

ageorgios, 2017-12-14:
I just received the EWelink 2ch Light Switch.
https://www.aliexpress.com/item/EWelink-2-Gang-1-way-Wireless-Wifi-Control-Light-Switches-Wall-Touch-Switch-App-Phone-Remote/32830632218.html
Not satisfied at all...
Buzzing sound, no serial connector holes, not reliable at all it is not working now...
better away from it.

gajotnt, 2017-12-14:
Glad i dint order that one... still waiting for mine to arrive.

https://www.ebay.com/itm/3-Gang-Wifi-Remote-Tempered-Glass-Panel-Smart-Touch-Wall-Light-Switch-EU-Plug/232558906972?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

petjek, 2017-12-14:
@kevinshane So the Sonoff T1 is bad quality? I was just about to order one for testing.

gajotnt, 2017-12-14:
The T1 is good quality, we are looking for alternatives

kevinshane, 2017-12-29:
@petjek the T1 is the best quality comparing to other "sonoff fake/like style x gang version"
so far I've bought at least 2 alternatives "sonoff like" switch and they all sucks, I mean really sucks!!!
stick to sonoff T1 bro~

ageorgios, 2017-12-30:
@kevinshane totally agree

gajotnt, 2018-01-16:
Just got mine, flashed with SonOTA, configured like a T1 3gang, working great so far.

https://www.ebay.com/itm/232558906972

slowfinger, 2018-02-04:
bought the same like gajotnt, but soldering was so bad that luckily before applying power it fell apart. Very dangerous !
[image]

LucaAghemo, 2018-02-17:
I got this 2 gang Sonoff clone from a friend of mine, who bought it on amazon italy and gave it to me because he was unable to put it in the wall, as Italian standard for switches is different:
https://www.amazon.it/gp/product/B076CJHPQ6/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1
It seems quite good to me, I kept it powered and connected (with standard ewelink) for some week, no heating problems, working properly.
Then I decided to open it, and see if I could flash it but... no luck. It has no reset button, the pin are forwarded on the PCB but... under the touch button 1!  I was able to reach them, solder some wires and try to flash it, but it was impossible to put it in flash mode. 
That said, the construction seems quite honest to me.
I was going to put it in closet (or in the trash bin) when I tryed a lsat, desperate thing, flashing it with SonOTA. Guess what: it worked! Almost...
Everything seem to work ok, mqtt connection, switching, I even connected it to Home Assistant. The only wierd thing is that the touch button 2 doesn't work anymore. The relais can be controlled via software, button 1 work perfectly, button 2... dead. This happen weather I power only the logic board with 3.3v or the whole thing assembled with 220v.
Quite a mistery to me, it used to work when standard, I tried 1 min before flashing, can't understand why.
Actually I'm the owner of a mix device, half T1 and half sonoff basic! Funny uh?
I'll try to resolve, and keep you informed. BTW the switch is not bad at all.

z4nD4R, 2018-03-01:
Hello All,

also looking for some alternatives for Sonoff Touch T1 - not because the lack of quality, but because of the fact they closed the OTA updates (yes I'm too lazy to solder every silgne wall switch :) ). 
So if anyone does know some alternatives with similar wall swtich easy to flash or already support MQTT, pls drop me message .. I'd appreciate it.

Y

LucaAghemo, 2018-03-02:
@z4nD4R, 
Kesen switch are not bad at all, apparently one gang and three gang are good with SonOTA, mine is a 2 gang and still give me some problem, as manual switch 2 doesn't work, but I think it might be a hardware problem. After all I did many test on it, so maybe it just suffered some ingiuries :-)
Switch one works properly, and relays 2 can be activated via mqtt, and both relays can be activated with an external switch, "a la sonoff basic", so I guess I'll save it in some way.
In Tasmota forum a guy wrote me that he succeded in flashing a 3 gang with Sonota, so I guess you can give it a try.
In case you do, please let me know.
Thank you.