arendst/Tasmota #1229: Need Help in setting up the Sonoff switch with Hassio

samiraa78, 2017-11-21:
Hi!

I have been banging my head for hours now cant find anything on the internet can anyone guide on how can i control the sonoff switch from hassio i have got the mqtt server configured on the switch and it detects on the mqtt server but post that what i need to do i am clueless

tom-niddatal, 2017-11-21:
are you talking about standard Sonoff of Sonoff-Tasmota?

samiraa78, 2017-11-21:
Sonoff-Tasmota

samiraa78, 2017-11-22:
Host (domus1)

192.168.0.99

Port (1883)

1883

Client (DVES_00BEE7)

DVES_%06X

User (DVES_USER)

DVES_USER

Password

•••••••••

Topic = %topic% (sonoff)

Masterlight

Full Topic (%prefix%/%topic%/)

%prefix%/%topic%/

The above is my settings in MQTT Server setup on the sonoff switch

Switch Masterlight "Masterlight" <switch> [ "Switchable" ]
 {mqtt=">[broker:cmnd/Masterlight/POWER:command:ON],>
[broker:cmnd/Masterlight/POWER:command:OFF]"}
                                                
The above is the text in my home item file 

sorry i seem to have no clue on what i am doing but i can turn on and off the device from MQTT fx but not for openhab2 

Can you guide me what i doing wrong

tom-niddatal, 2017-11-23:
I would suggest to use the MQTT Broker within Home-Automation. If so you should initialize this using some code like this in configuration.yaml:

# Tracks MQTT
mqtt:
  broker: 10.0.0.64
  port: 1883
  client_id: home-assistant-1
  keepalive: 60
  username: HAUSERNAME1
  password: HAPASSWORD1
  protocol: 3.1
 
Using MQTT as sensor use the following code in configuration yaml - sensor section:

 MQTT
  - platform: mqtt  
    name: "LED-Sonoff"
    state_topic: "stat/sonoff/POWER1"
    paylod_on: "ON"
    payload_off: "OFF"

Using MQTT as switch use the following code in configuration yaml - switch section:
  - platform: mqtt  
    name: "LED-Sonoff"
    state_topic: "stat/sonoff/POWER1"
    paylod_on: "ON"
    payload_off: "OFF"

Check out which topics you want to use from MQTT. The ones I have listed are standard from Tasmota.
In Addition you can use telemetric data from Sonoff. Could would be sth. like for Wifi Info i.e.:
- platform: mqtt
    name: "Time"
    state_topic: "tele/sonoff/STATE"
    value_template: "{{ value_json.Wifi.SSId }}"
    unit_of_measurement: ""
  - platform: mqtt
    name: "RSSI"
    state_topic: "tele/sonoff/STATE"
    value_template: "{{ value_json.Wifi.RSSI }}"
    unit_of_measurement: "dB"

Hope this helps you somewhat further.
Tom

samiraa78, 2017-11-23:
Thanks Tom but got i working the commands i was using in openhab were all wrong

closed 2018-05-07 by stale[bot]