arendst/Tasmota #1230: Extend sleep duration above 250 ms

roblad, 2017-11-21:
Hi,

Is it possible to prepare ease set deepsleep mode for battery saving as it is developed in:

https://www.losant.com/blog/making-the-esp8266-low-powered-with-deep-sleep

"""
Hi @davidelang,

the link is here: https://github.com/stefanbode/Sonoff-Tasmota/tree/master/sonoff

Additional features are

DeepSleep support up to unlimited (deepsleep=xxx [sec]), support also 1day or more deepsleep
Improved Startuptime incl TLS MQTT and measurement in <9 sek
Support for Ultrasonic distance measurement HC-SR04
Support for I2C 10bit 4-channel A/D converter ADS1115
Support for I2C 8-channel DIO extension board PCF-8574 (large extension to support 32 relays and pulsetimers on 32 relays; other functions still stick to 4)
Report total UPTIME (seconds) between two deepsleep cycle. This allows to see uptime, even if connect was not successfull or there was a reboot. Using RTC memory to ensure correct uptime
Support UtraDeepSleep on any duration in seconds. Just define deepsleep e.g. 1 day = 86400. (wakes up every hour for 0.4 seconds until time reached)
Regarding button presses, there is no chance to capture this during deepsleep. This functionality comes with the ESP32 but is not available with the ESP8266. MQTT messages (Power switch) are partly working. As long as there is only ONE relay and the message is retained after wakeup the status is updated. It seem a limitation of ESP8266 and/or PubSubClient to transmit only the first retained message on a wildcard subscribe. I have checked several libraries but did not get it working at least. A little bit pain-in-the-... , because there where much more ideas possible if you can work on ALL topics that came from retained messages after a wildcard subscribe.
My use-case is "measuring". E.g. moisture, water-level and so on. This works fine.
"""

arendst, 2017-11-22:
Sorry won't implement. Not the core business of Tasmota.

ronnicek, 2017-12-17:
Why there is not an option to extend it for example 5 / 10 / 15 minutes? :)

smadds, 2018-03-01:
I'd like to add a +1 to this request, Theo. I understand the core is Sonoffs, but this firmware is now very flexible and I would like to use it in as many ESP8266 control applications as I can. 

I'm currently looking at wifi enabling a smoke detector. It would be ideal if I could wake the processor up every hour to log battery volts as well as using the alarm detection to trigger a reset. 

My understanding is that the ESP has a max sleep time of about 71 minutes - would it be difficult to least allow the sleep time to extend to that?

roblad, 2018-03-01:
I am working on it I have ADC measure and I am juzt preparing MQ-5 gas sensor.