arendst/Tasmota #1231: Sonoff POW won't reboot

MatsKarlsson, 2017-11-21:
Successfully flashing in DOUT, 512 (no spiffs) etc. then after it reaches 100% it starts up.

Then when I change some settings and saves it reboots and I get boot mode: (1,7)
Isn't the 1 that it should boot from UART? 
Shoudn't GPIO0 be correct (ie. wired to 3.3v)?

```
...
20:44:06 MQT: stat/pow1/RESULT = {"POWER":"ON"}
20:44:06 MQT: stat/pow1/POWER = ON
20:44:09 MQT: stat/pow1/RESULT = {"POWER":"OFF"}
20:44:09 MQT: stat/pow1/POWER = OFF
20:44:10 MQT: stat/pow1/RESULT = {"POWER":"ON"}
20:44:10 MQT: stat/pow1/POWER = ON
20:44:14 MQT: stat/pow1/RESULT = {"POWER":"OFF"}
20:44:14 MQT: stat/pow1/POWER = OFF
20:44:44 OTH: MQTT enable ON, Emulation 0, FriendlyName pow1, Sonoff2, Sonoff3, Sonoff4
20:44:46 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

ionciubotaru, 2017-11-22:
it happends, disconnect uart and it will boot

arendst, 2017-11-22:
512 (no spiffs) is no good. You will need 1M (no spiffs) for a pleasant experience.

MatsKarlsson, 2017-11-22:
Thanks @arendst , That did the trick! now it works. 
Missed the line on Upload-page where it said: **If Version 5.x.x -> Flash Size: "1M (no SPIFFS)"**

Feeling so stupid, sorry for taking your time.

closed 2017-11-22 by MatsKarlsson

nixmeer, 2017-12-24:
Hello :) 
Arduino flashes my Sonoff Basic when choosing "NodeMCU 1.0 (ESP12-E Module)" but Sonoff is dead afterwards. It does not flash when "Generic ESP8266" is selected, maybe because Arduino does not offer 1M(no Spiffs). Any idea why?

Best 
Jan