arendst/Tasmota #1232: unit going on and offline constantly

ins4neuk, 2017-11-21:
Unit gives out the online mqtt command and then within a few seconds then says offline. tried to access the phone wireless and doesn't work but does work with a manual switch on the unit which i installed.

Any suggestions?

ionciubotaru, 2017-11-22:
Improve the wifi signal

pascalsaul, 2017-11-22:
I've the same with a basic and I tried everything. Reset 2 and new firmwares and it keeps going online and offline everytime. Sometimes with a exception but can't figure out what triggers it. I just replaced it with an other module now. The AP was within 2 meters in my case ;)

ionciubotaru, 2017-11-22:
If your RSSI is better than 60% and sonoff still disconnects, try to change the AP channel, maybe is another AP running on the same channel

MrPasty, 2017-11-22:
I too have just installed this on a sonoff S20 and mine is doing the same, i have two other sonoffs running older firmwares and both are rock steady - this one keeps turning off and on and the web interface locks up after connecting just once

MrPasty, 2017-11-26:
Did you manage to fix your issue? its unusable for me, both the minimal and the full version - just drops off the net constantly - even when its beside the wifi AP!

pascalsaul, 2017-11-26:
Nope...  can try to flash it with 5.9.1i. Also waiting for new FTDI based on TI chip. My RSSI was 100% also.

chiefymuc, 2017-11-26:
I have a similar issue, but only when 2 devices are online. As soon as I cut the power to one of them, the device is stable.

Is there maybe a condition where two device can powercycle each other constantly?

The naming is quite different: sonoff-H801-1 and sonoff-H801-2

MrPasty, 2017-11-26:
I tried 5.9.1i and same thing, i also have two FTDI and tried them both.  

My other S20 and standard sonoff are running 5.3.1 and rock solid - i tried to put 5.3.1 on this one but doesn't seem to work - I think its due to the flash method and i cant find the old settings i need to use, perhaps i just need to totally wipe it and try again, even though i have done erase_flash on esptool.

I haven't tried turning off the others to see if they are interfering but i find the whole thing hit and miss, webservers unreachable most of the time etc Its so frustrating tempted to buy official wemos and be done with it.

davidelang, 2017-11-26:
check the logs of both of them.

also check the logs on the mqtt server, there are multiple places for devices to 
be named, and if they are conflicting, that can cause grief

ins4neuk, 2017-11-27:
i think i narrowed mine down to the IP address. i think some how my router was dishing out the same ip to both devices which is why it was killing it. i managed to get one up forget the details for it and then it grabbed a different IP and bingo were sorted

MrPasty, 2017-11-27:
Logs on my other devices are clean and definitely not a duplicate IP cause I've tried several.  I think the unit is just rebooting because it resets the time to 0:00 and shows this in the log after it's booted and corrected the time:

21:00:10 MQT: tele/LOUNGELAMP2/INFO3 = {"RestartReason":"Blocked Loop"}
21:00:11 UPP: Multicast (re)joined

ionciubotaru, 2017-11-28:
Try to move your dhcp server on a raspberry, because routers dhcp cannot handle lots of clients - in my case this solved my problem.

MrPasty, 2017-11-28:
I am using a static IP so i dont think thats my issue, thanks for the suggestion though

GAZ082, 2017-12-11:
@MrPasty same issue, Blocked Loop. Have you figured out? Worked perfectly with HA just yesterday.

MrPasty, 2017-12-13:
Nope still does it, seems to randomly reboot - if i access the web gui it breaks straight away, if i just control it with mqtt it works for a while but randomly reboots still

GAZ082, 2017-12-13:
@MrPasty i had a issue with the Sonoff rednode node. Stay away from it and keep using the mqtt node.

MrPasty, 2017-12-13:
@GAZ082 Sorry what do you mean rednode node? im just using MQTT on the Sonoff pointing to my Mosquitto server?

webtop, 2017-12-26:
I too am now having this issue. Everything was fine 2 days ago, now however, the unit reboots before being fully initialized.

Output from the device is
```
--- Miniterm on COM3  115200,8,N,1 ---
00:00:00 Project sonoff Sonoff-Kitchen-Nook (Topic nook-sensor, Fallback SONOFF_KITCHEN_NOOK, GroupTopic sonoffs) Version 5.10.0
00:00:00 WIF: Connecting to AP1 ALLSOPP-2-4G in mode 11N as nook-sensor-2510...
00:00:06 WIF: Connected
21:20:19 DNS: Initialized
21:20:33 HTP: Web server active on nook-sensor-2510.local with IP address 192.168.0.25

ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
```

I'll try a diff device to see if the issue is H/W, S/W, or BKAC

webtop, 2017-12-26:
Indeed, in my case it seems the module itself is failing, so I'll have to erase it completely and then reflash it.

closed 2018-05-09 by stale[bot]