arendst/Tasmota #1235: switch is missing S0 impulses with 35ms length

MatzeJoerling, 2017-11-22:
Hi,

have tasmota on esp-12f running, my power meters sending 800Imp/KWh. 
I have a switch gpio attached to S0 output (pull down resistor) and tasmota is counting every impulse and sending it to my mqtt server (there i have attached node-red and influxdb für stats and counting). 
Every Impulse, no ... 10% are missing. I am attaching a screenshot from the impulse. 
it is about 35ms long (acording to wikipedia S0 has to be 30 to 120ms) 

i have a switch im mode 3 (toggle), Switchtopic set. 
any idea why some 10 to 15 % of the impulses are missing?

[image]

arendst, 2017-11-22:
Why on earth are you using switch for this? Try a counter as it is interrupt driven where the switch is just polled every 50 mSec.

MatzeJoerling, 2017-11-22:
I want to have "real time" statistics, power meter should show the actual value. 
if you say polled every 50mSec, can i compile with other values? can you tell me where to tune the poll time?
what about the possibility to use a interrupt for a switch? as an option?

davidelang, 2017-11-22:
well, the power meter is always going to lag behind a bit, but if you use the 
counter module and a 10 second teleperiod, you will get updated every 10 
seconds, which is pretty close to current

the polling time is probably defined in sonoff.ino if you want to start 
tinkering.

MatzeJoerling, 2017-11-23:
Thanks @davidelang,

that was really not much information but i also have to deal with "users" and know how to encorage them to search and learn... but anyway.

I found out that the base loop timer is 50ms. So no easy way to get "faster" switches.
What i did is double the STATE from 20 loops per Second to 40.

`
#define STATES                 40           //  was 20 / State loops per second
`
needless to say that i have to fix all other timings in sonoff.ino (0.1s/0.2s ...)

After some tests i think i fixed my problem. 
But i would like to discuss this as **feature request!**
**Please tune up the base loop to 25ms (40 loops per second).**
I home / think esp can handle this?? if you want to messure fast real time S0 impulses you need sample freq of 1/25ms.

MatzeJoerling, 2017-11-23:
Hi, as seen on the screenshot (old green, new 25ms loop orange) the new one gets almost all impulses, the green (old) is missing some, more impulses missing with less load (every impulse is 1/800 KW/h).

[image]

Please consider changing the loop time to 1/40s .

regards

arendst, 2017-11-23:
The timing of 50mS is choosen to fix key/switch debounce. Again, you should select an interrupt driven  counter as switches will need this 50mS debounce time.

MatzeJoerling, 2017-11-24:
ok, no switches. but i need every impulse and i also dont want to pollute my network with telemetry every 10sec. 
what about some counter enhancements like CounterMode 3 (send every impulse) and CounterTopic .
Every Impulse is send to specific mqtt topic to cmnd/coutertopic/counter[1234]
I think many people need some "real time" measurements like that.

davidelang, 2017-11-24:
you don't want to pollute your network with a message every 10 seconds, but 
instead you want to send a message for every pulse (which needs to be set to 
check every 25ms or faster to avoid missing one)???

This doesn't compute. especially as you admit that even this shorter time still 
misses some pulses.

you need an interrupt driven counter, not use switch detection. figure out how 
frequently you need updates and settle for that, "real time reporting" and 
"don't pollute my network" don't go together.

David Lang

MatzeJoerling, 2017-11-24:
pollute means the status message contains ALL data (status,temp, wifi..) and 2Byte counter. The overhead is really big compared to 1 message per impulse with 100% wanted payload. 

Again, i am on my way to interrupt counter, but want to have to option to send every impulse.

MatzeJoerling, 2017-11-24:
Hi, 

switched to counter mode and made some tests, indeed esp gets 10ms impulses very acurate. 
while doing that i saw that you update the website in realtime?! 
I took a look at the counter / mqtt code but that is far beond my skills.

If you update the webpage in realtime, could you please consider to add an option to upate mqtt also?
What do you say? 

regards M.

MatzeJoerling, 2018-04-26:
Hi,

again i want to raise this feature request.
please implement toe option to send every counter impulse to mqtt. You have this on the webpage updating counter nearly in realtime.
this is indeed only for a small used case where telemetry time is much more than impulse time and certainly impulsetime is not smaller than 1/5s or so.

regards M,.

closed 2018-06-25 by stale[bot]