arendst/Tasmota #1236: Support for WeMos D1 ESP-Wroom-02 Nodemcu ESP8266

roblad, 2017-11-22:
Hi,

Do you plan to support

http://espressif.com/en/products/hardware/esp-wroom-02/overview

https://www.ebay.com/itm/WeMos-D1-ESP-Wroom-02-ESP8266-Nodemcu-WiFi-Module-With-18650-Battery-Charging-/192211314504

davidelang, 2017-11-22:
I see no reason why that shouldn't be supported now. We don't take advantage of 
the extra flash (but we don't take advantage of the extra flash in other modules 
as well) but other than that it should act just like any other ESP8266. (or am I 
missing something in my quick look through the specs??)

roblad, 2017-11-22:
Hi,

What template will work on it, I now that it has 2M  but I do not need that 1M enough :-) 
http://espressif.com/en/support/download/documents?keys=&field_type_tid%5B%5D=16

could you check it out, and if you can advice me if WeMos D1 mini",   // WeMos and NodeMCU hardware (ESP8266) will work on it. Additionaly in my previous issue I send a question to you about deepsleep - it is because of that :-) GPIO16 is used for wakeup from deepsleep and it can save an energy. It has also Voltage divider for control voltage, could be also charged i.e. by the solar baterry - it is very nice solution for nodes for measurment IOT etc.

davidelang, 2017-11-22:
the different module types are based on what is wired to what GPIO in that 
module.

The wemos allows every GPIO to be user defined (it really should be called 
generic or something like that), and so it will work on anything.

The question is what do you wire to this module? that's what matters.

roblad, 2017-11-22:
Hi,

Maybe you could add deep sleep for that one if you can ?

sebul0n, 2017-11-26:
I'm also interested to get working deep sleep on "WeMos D1 ESP-Wroom-02 18650". In arduino IDE on "ESP8266WiFi.h" there is a deepsleep-funciton, but in order to get that working GPIO16 and reset should be wired together? Am I right? However..in this board the GPIO16 is used for a LED?

Please let me know if there is a chance to get deep sleep working on this board?

roblad, 2017-11-26:
Hi,

I am waiting, I ordered some, when I will have it I will see. Probably the LED not disturb that there is on board one place where you ned solder 2 pins for deep sleep. Do you have schematic of that ?

sebul0n, 2017-11-27:
Hi Roblad,

You were totally right! I found the place where I soldered two "pins". I marked the place with red color in attachment. Thank you very much, the deep sleep is now working!=)

[image]

pepcampas, 2018-01-26:
Hi,
i can't find anywhere the pinout for this board. Does anybody knows wichs pins are SDA and SCL (from D0 to D8) ?
Thank you

mkh595, 2018-01-29:
From Google:
[http://www.etteam.com/productSensor/ET-SENSOR%20BME280/ESP8266_BME280_I2C_ADAFRUIT_LIBRARY.pdf](http://www.etteam.com/productSensor/ET-SENSOR%20BME280/ESP8266_BME280_I2C_ADAFRUIT_LIBRARY.pdf)

[https://www.ei.tohoku.ac.jp/xkozima/lab/espTutorial3.html](https://www.ei.tohoku.ac.jp/xkozima/lab/espTutorial3.html)

[http://skomo.o.oo7.jp/f49/hp49_17.htm](http://skomo.o.oo7.jp/f49/hp49_17.htm)

pepcampas, 2018-01-30:
definitely are pins 1 and 2 (gpio 4 and 5).

closed 2018-07-10 by ascillato2

livello, 2019-06-29:
The problem with that module is power consumption in deep sleep mode. It has usb-ttl converter and cheap 3.3v diode. Both of them eat current and device works two-three days on 18650 3000mah....

cmgreenman, 2019-10-26:
> The problem with that module is power consumption in deep sleep mode. It has usb-ttl converter and cheap 3.3v diode. Both of them eat current and device works two-three days on 18650 3000mah....

I was able to get one to last for 24 hours using just by forcing modem sleep and waking every 20 seconds to send an mqtt message.  This was using a Samsung 2500mah 18650.  I'm hoping to use this coupled with a modest, 1A usb solar panel (Renogy) to drive my office blinds.  If I can get satisfactory results with the setup then I will hookup the servos and see what I get. My limited current reading capabilities (analog output meter on a cheap chinese 3 in one solder station/power supply) show in line with the rated specs of around 17ma.  I'm hoping to get deep sleep to work but when I tried shorting the pads as mentioned above, the device wouldn't boot.

KDMcMullan, 2019-12-17:
The worst feature of the "WeMos D1 ESP-Wroom-02 18650" is that the "left" button is wired to RESET. Has anyone got a workaround to rewire this to a GPIO instead?
Should I be able to flash this unit with the TASMOTA binaries as they are?

pkejval, 2019-12-18:
> The worst feature of the "WeMos D1 ESP-Wroom-02 18650" is that the "left" button is wired to RESET. Has anyone got a workaround to rewire this to a GPIO instead?

Just look above: https://github.com/arendst/Tasmota/issues/1236#issuecomment-347125849

Yes, you can flash Tasmota binaries as they are. Template Generic and you are ready to go. :)

KDMcMullan, 2020-01-17:
> Just look above: [#1236 (comment)](https://github.com/arendst/Tasmota/issues/1236#issuecomment-347125849)

I can't see teh connection between your response to my statement that the "left" button is connected to RESET.

haldi4803, 2021-07-29:
> 
> 
> Hi Roblad,
> 
> You were totally right! I found the place where I soldered two "pins". I marked the place with red color in attachment. Thank you very much, the deep sleep is now working!=)
> 
> [image]

this has to be the only image on the web properly afirming that this is the correct solder point! Thanks.

Blak-DK, 2022-07-27:
How do i flash tasmota on this, i have flashed many other devices but not tis one :-)

ozett, 2023-01-31:
anybody success with this module and the webinstaller today?
uart is found, but will not flash.. do i have to do it with esptool?

appreciate any hints.. thanks

aagilang, 2025-02-02:
i got this node mcu with broken usb, how to flash using usb ftdi ? which pin do i use ?

lalo-uy, 2025-02-03:
You need to hookup the ftdi to tx & rx pins.
But the broken usb adapter is also connected there, so it may interfere.